FreeCAD/FreeCAD #8884: [Problem] C++ DXF importer never imports text (annotations)

kpemartin, 2023-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32303 (Git)
Build type: Debug
Branch: master
Hash: 45ee407645c147147656ec63cec9da5674516493
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

The C++ DXF importer no longer imports text from the DXF file. This worked fine in the 0.20 importer, and still works fine with the legacy importer.

To reproduce, import the attached DXF file (contained in [Missing Text.zip](https://github.com/FreeCAD/FreeCAD/files/10970418/Missing.Text.zip)) with preferences: Do no use the legacy importer, import text and dimensions. Notice that the text in the file does not get imported. To compare, import with the legacy importer.

In fact, the C++ DXF importer ignores _any_ choices in the Preferences -> Import/Export -> DXF dialog; you are stuck with No texts and dimensions, do not group layers into blocks, scaling factor 1.0 (the only settings that seem to make any difference)

The problem is that this dialog places the preferences in `User parameter:BaseApp/Preferences/Mod/Draft` but the code loads the settings from `User parameter:BaseApp/Preferences/Mod/Draft`.

More specifically, the `ImpExpDxfRead` class ctor reads settings from `User parameter:BaseApp/Preferences/Mod/Draft` (finding the correct settings), but then method `readDXF` in class `Import::Module` in file AppImportPy.cpp forces settings to be loaded from `User parameter:BaseApp/Preferences/Mod/Import`. Because of the way `ImpExpDxfRead.setOptions` works, you get the class's default settings as modified by whatever is in `User parameter:BaseApp/Prefernces/Mod/Import` and the previous settings from `User parameter:BaseApp/Preferences/Mod/Draft` are discarded. Then because `User parameter:BaseApp/Preferences/Mod/Import` actually contains no settings, you always get the class's defaults which are: No text/dimension import, no grouping layers into blocks, scaling factor 1.0.

The problem started with [this commit](https://github.com/FreeCAD/FreeCAD/commit/e1ebcef39e8f53ada00ad77ba800a0132aef61e8) which changed `readDXF` to get the settings from Mod/Import. There were no corresponding code changes along with this commit to change the legacy DXF importer to look in the new location, to have the Preferences dialog place the options in the new location, nor to migrate the settings from the old place to the new one.

The easiest fix would be to roll back  [that commit](https://github.com/FreeCAD/FreeCAD/commit/e1ebcef39e8f53ada00ad77ba800a0132aef61e8).

The next easiest fix would be to change `ImpExpDxfRead::setOptions` so that any previous options (rather than canned defaults) are used as the basis for changes. Thus options that do not exist at all in Mod/Import would have their settings from Mod/Draft. This is the change I've made in my local copy of the source code and can supply it if that seems to be the preferred change.

The hardest fix would be to completely migrate this group of settings from Mod/Draft to Mod/Import; this requires changing the legacy DXF importer to look in the new location, changing the `preferences-dxf.ui` file to place the settings in the new location, and to add some code (I'm not sure where) to migrate saved settings in the user.cfg file from the old location to the new one.

My preference is for the first fix (roll back). The second fix is easy, but it still supports this orphaned migration of the settings to a new location that is not otherwise supported. The third fix  seems to be to be the "proper" fix as I feel Mod/Import is a better place for the settings, but it feels like a lot of work to make a change merely on principle which has no net effect on the operation of the program.

It might also be useful to clean up which code is responsible for loading the settings. Right now, the C++ body for `readDXF` actually accepts `(filename, documentName, ignore_errors, optionSource)` as a Tuple, with the documentName specifying the document to place the objects into (default to the current document), ignore_errors defaulting to false, and optionSource defaulting to .../Mod/Import (with the bad commit). But the ctor for Module registers the readDXF method in Python only as `(filename, [document, ignore_errors])`, not documenting the ability to pass an alternative settings group. I just think there are too many cooks in the kitchen here; if no optionSource is passed to `readDXF` it should not be trying to set options on the `ImpExpDxfRead` object at all, relying instead on the defaults this class has already loaded.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-15:
CC @wwmayer as this ticket references https://github.com/FreeCAD/FreeCAD/commit/e1ebcef39e8f53ada00ad77ba800a0132aef61e8

kpemartin, 2023-03-19:
I have a fix for this and also #8806. I started writing a draft PR for it but found the branch was contaminated with other changes (in particular, from another PR that I have still pending) so I have to clean up the mess before submitting the new PR.

wwmayer, 2023-03-22:
https://github.com/FreeCAD/FreeCAD/pull/8997

donovaly, 2023-03-24:
Is this fixed by #8997?

kpemartin, 2023-03-24:
The problem visible at the surface, that text fails to import, has been fixed by #8997 but PR #8959 is still active because it addresses underlying problems in the options handling code which are partly the cause of the problem. Issue #8806 originally related to this as well but has been amended and now refers to a separate issue.

wwmayer, 2023-03-24:
> What is even more interesting about the export side is that the C++ exporter looks for settings called 

It's not uncommon that not every single parameter has a control widget in the preferences. Some less important parameters can only be changed with the parameter editor. That's not only the case here but elsewhere, too.

> It looks like the C++ exporter is completely out of touch with the location of the options, or even what the options are.

It's probably best to harmonize this so that both readers/writers access the same parameter group.

donovaly, 2023-03-24:
The bugfix will be available in FreeCAD 0.20.3 -> closing

closed 2023-03-24 by donovaly

yorikvanhavre, 2023-03-27:
The whole mess with the DXF preferences is because at the beginning, the C++ DXF importer was part of Draft, and was meant to replace the Python importer (which never fully happened). So we began implementing some of the preferences options. Later on, it was moved to the Import module, and nothing was done regarding preferences.
Probably the best way forward would be to indicate in the DXF preferences UI which settings affect the C++ importer...