FreeCAD/FreeCAD #8886: [Issue] transform tool

colt45s1911a1, 2023-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CADExchanger
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.2.59
  * toSketch 1.0.1
[/code]
```

### Subproject(s) affected?

None

### Issue description

i can transform objects, but they won't stay.
once you click OK it returns to the position it was in before it was transformed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-15:
Is this a workflow regression and are you sure you're initiating the workflow correctly ?

colt45s1911a1, 2023-03-15:
big words confuse me, i'm not sure of the meaning of "
workflow regression", i'm simply clicking on transform after selecting a
body and then moving it, i click ok and the movement is then undone, back
to where it started like i have done nothing.
I attached the file, there are 3 different bodys merged, none will stay
transformed.
I'm pulling the 2 tubes away from the base to give a better view of the
assembly.

colt45s1911a1, 2023-03-15:
UPDATE: if i choose the from the edit drop down menu rather than the right
click menu when selecting transform it does not revert my changes back to
where they started.
i also double checked that if i right click and select transform it will
still revert back once i click ok.
i hope this helps.

colt45s1911a1, 2023-03-15:
adding to this, once i used the right click menu to select transform the
edit drop down menu began doing the same reversion.
I closed the file and then reopened it, once i did the transform under the
edit menu worked as it normally should again.
This didn't require me to close FC, only the file I had open..

colt45s1911a1, 2023-03-16:
[piston seal cutter comp.zip](https://github.com/FreeCAD/FreeCAD/files/10993821/piston.seal.cutter.comp.zip)

Syres916, 2023-03-18:
I'm not sure whether it's a genuine mistake but please attach the **_exact_** FCStd file you are using not something that looks like it's come from Thingiverse, it will make the devs job a lot easier.

colt45s1911a1, 2023-03-19:
that is a FC std file. i can send the 3 individual files if you don't believe me i can add them separately, they are just merged together, it's pointless to show the problem with transforming if you don't have a reference. 
regardless i have been troubleshooting this if you read previous comments it should give some insight.

Syres916, 2023-03-19:
Well the file doesn't open on my system and when I extract it there's no Document.xml which is the fundamental part of the file spec. I've read your comments and cannot replicate any of the issues on 0.20.2 nor 0.21usinng today's master.

colt45s1911a1, 2023-03-21:
syres916 did those files help?

Syres916, 2023-03-22:
I'll look at this tomorrow when I've some more time.

Syres916, 2023-03-24:
With the 3 files merged into one, I've tried adding all the same addons that you have (toSketch was of particular interest with it's Reset to Origin function) but I just cannot replicate the issue. My only suggestion is (with FreeCAD closed) to **_rename_** your `user.cfg` and `system.cfg` files in `C:\Users\`_username_`\AppData\Roaming\FreeCAD` to something like `user.old` and `system.old` and rename the `Mod` folder to `ModOld`.
Start FreeCAD which will be like a new user would see, open the file and retry. If it works as expected then install just _one_ of your addons at a time and keep testing after each one to see the issue reappears.

If you've a forum user account then you can private message me a copy of just the user.old file (it's possibly got information that you won't want in a public place) and I can go through alongside my own Windows setup to see if I can find the culprit but it's a long job.

adrianinsaval, 2023-03-28:
two possibilities I can think of: you are launching the transform tool from 3d view, this selects the individual PD features (which can't be moved) not the body, so the transformation is discarded. Or in your assembly you have attached the bodies to something, in which case the transform is again discarded.

colt45s1911a1, 2023-03-30:
i appreciate the help with this, but my fix was to install FC19.4
everything works and im having no other issues.
i'm just going to wait for the bugs to get worked out of FC 20.

adrianinsaval, 2023-03-30:
I suggest closing this issue then since no bug has been identified and OP is not providing further information

colt45s1911a1, 2023-03-31:
a bug not found doesn't mean a bug doesn't exist.

closed 2023-03-31 by colt45s1911a1

adrianinsaval, 2023-03-31:
yeah, but if you don't give us a reproducible case what are we supposed to do? I explained certain cases where it is normal that this happens and it's unclear if this is what you experienced or something else. Without more info there's nothing to do.

aperezhra, 2023-08-28:
I'm not sure if it can be of any help. But for those who are asking for a steps to recreate, here are some that I was able to replicate in the 0.20 and 0.21 versions:

**1. Body#1 (for reference)**
2. Create a circle sketch
4. Pad the sketch, 5mm
**5. Body#2 (to Transform)**
6. Create a link from sketch from Body#1
7. Create a sub-object shape binder from the linked Sketch
8. Pad the binder sketch, 10mm
9. Right-click on the tree view over the Body#2, select Transform
10. Move the Object to a new position on the X or Y axis
11. Click Ok
12. The Body#2 returns to its original display position, even though the Body's Placement Position has been changed.

[image]

colt45s1911a1, 2023-08-28:
Thank you for helping. Sent from my iPhone

adrianinsaval, 2023-08-30:
that is completely expected, a subshapebinder tracks the position of the binded object so even if the body is moved the binder will remain in the same place. If you want a movable copy of the sketch I suggest using a carbon copy instead