FreeCAD/FreeCAD #8887: [Issue] export DWG problems

colt45s1911a1, 2023-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CADExchanger
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.2.59
  * toSketch 1.0.1
[/code]
```

### Subproject(s) affected?

None

### Issue description

when i open exported DWG files they are distorted and don't resemble the drawing, this is new, in previous versions this didn't happen. im unable to attach any files to show as it wont let me attach FC files or DWG. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-03-15:
We need more info here, basically:

* Your DWG file (if you zip it, you can add it to this issue)
* Which convertor you are using (In FreeCAD, under menu Edit -> Preferences -> Import/Export -> DWG)

You might also try to choose one convertor in the menu above, and set the path accordingly.

I am just noticing that the error message you get from the DXF to DWG translation is unclear. If the convertor is set to "Automatic", it will give a "QCAD error" when actually no valid convertor has been found. I'm fixing that now...

luzpaz, 2023-03-15:
> im unable to attach any files to show as it wont let me attach FC files or DWG.  

Please zip the files and drag drop them in to the github text box.

colt45s1911a1, 2023-03-15:
I am using the std export from the drop menu, I did have CAD converter
installed. I uninstalled it and performed a restart but nothing changed.
I also used a different program to open the exported file to see if it was
opening that file type problem, it was not, drawing was the same in the
other program as it was in FC.

colt45s1911a1, 2023-03-16:
[AIR cyl mount plate tab.zip](https://github.com/FreeCAD/FreeCAD/files/10993828/AIR.cyl.mount.plate.tab.zip)
[AIR cyl mount plate tab.zip](https://github.com/FreeCAD/FreeCAD/files/10993830/AIR.cyl.mount.plate.tab.zip)

colt45s1911a1, 2023-03-16:
[image]
[image]
this is after a complete uninstall, scrubbed clean the previous instal.
reinstalled and restarted, sill doing it, DXF and DWG.

luzpaz, 2023-03-18:
> Which convertor you are using (In FreeCAD, under menu Edit -> Preferences -> Import/Export -> DWG)

Please respond to yorik's question

yorikvanhavre, 2023-03-20:
One of the zip files contains a FCStd file, and the other a DXF file... Could you add the DWG file instead?
The DXF file, when I open it in QCAD, shows the same thing as in FreeCAD. I have no AutoCAD or any other DWG app at hand unfortunately... Could it be that it contains solids? If yes, unfortunately I think CADConverter is the only one able to handle DWG solids correctly... The other ones (QCAD and LibreDWG, which is IIRC bundled with the Windows installer of FreeCAD, so it is likely what you are using now), won't handle solid objects.

This would explain why it worked before for you and now not anymore...

colt45s1911a1, 2023-03-20:
i am unable to export dwg, i get a Qcad error that is new, also my bad on the dxf or dwg, when i look in the folder its file type is listed as a dwg when it is actually a dxf.
im disappointed it worked for you, i have opened that same file on sendcutsend.com website and it is just as much a mess as when i open it.
as for the bundled programs for this in the windows installer, this was not a problem until the last update.
below is a file i made 6 or so months ago still opens fine. 
[METAL DETECTOR GUARD BASE-Body.zip](https://github.com/FreeCAD/FreeCAD/files/11020648/METAL.DETECTOR.GUARD.BASE-Body.zip)

colt45s1911a1, 2023-03-21:
so i pulled out an 
old laptop running the version 19.
opening the previously exported file it opens the same as it does in the current version. i then used v19 to export the FC file created using v20.2, same problem exporting dxf is not good.
so i redrew the piece using v19 and exported that, no problems works as expected.
i'm attaching all the files as i don't know what i'm looking for but something in the FC files must be different. 
both file labeled unnamed were created on the v19 FC. 
[AIR cyl mount plate tab.zip](https://github.com/FreeCAD/FreeCAD/files/11033421/AIR.cyl.mount.plate.tab.zip)
[unnamed 1.zip](https://github.com/FreeCAD/FreeCAD/files/11033422/unnamed.1.zip)
[Unnamed1-Body.zip](https://github.com/FreeCAD/FreeCAD/files/11033423/Unnamed1-Body.zip)

colt45s1911a1, 2023-03-22:
to speak to your idea of using CAD exchanger, it was installed with my 20.2 v it wasn't working, i uninstalled it didn't help.
i do have CAD exchanger installed with v 19. 
i don't think i was clear in my telling of the last post, using a flash drive i opened a FC file drawn on v20.2 using my v19. it opened fine then i exported it via v19 as a dxf. when i tried to open it it was distorted. 
then using v19. i redrew it then exported it as a dxf. when i opened it it was just as it was drawn, those files are attached in my previous response. 
if it was the exporter then it would work no matter what version the drawing was drawn with, this is a problem with the FC file drawn in v20.2 that is why it couldn't be exported in v19 correctly as well as in v20.2. 
i will look for links to the other older versions to find where the change happened, but that is easier said than done.

colt45s1911a1, 2023-03-22:
this is new, i received this error message after i installed and was setting up v19.4.
hope that is helpful i don't fully understand what its saying yet it seem pertinent 
[image]

colt45s1911a1, 2023-03-30:
i appreciate the help with this, but my fix was to install FC19.4
everything works and im having no other issues.
i'm just going to wait for the bugs to get worked out of FC 20.

luzpaz, 2023-04-01:
What's left to do in this ticket ?

colt45s1911a1, 2023-04-03:
figuring out the problem would be cool, going backward is not the way forward if it was unclear to you. 
i've experimented and tried several different thing to find a way around, the problem and give as much insite as i can.

yorikvanhavre, 2023-04-07:
For me the problem is clear (unability to export solids to DXF/DWG) and there is unfortunately not much we can do about it, at least now. We should nevertheless have a look at libreDWG and easyDXF, check if any of them supports ACIS/SAT solids. If yes, there might be something to do.

I propose to leave this open, as a reminder..

luzpaz, 2023-04-08:
> For me the problem is clear (unability to export solids to DXF/DWG) and there is unfortunately not much we can do about it, at least now. We should nevertheless have a look at libreDWG and easyDXF, [...]

Anyone

yorikvanhavre, 2023-04-11:
ezDXF has a partial support of ACIS solids: https://ezdxf.readthedocs.io/en/stable/tools/acis.html Probably not enough for our purposes...

LibreDWG apparently supports ACIS solids now: https://savannah.gnu.org/forum/forum.php?forum_id=9791 Also worth noting that LibreDWG is still actively developed: https://github.com/LibreDWG/libredwg

Now, the way we use LibreDWG in FreeCAD is simply to convert DWG files to DXF, and then importing them with the DXF importer. Same thing when exporting. This will not solve the ACIS issue as the ACIS blobs will still be present and untouched in the DXF file.

LibreDWG now has Python bindings. This could be used to do something... I'll investigate further

Roy-043, 2023-08-21:
There is some miscommunication here. AFAICT the issue is not about exporting ACIS solids.

The difference between the supplied files, https://github.com/FreeCAD/FreeCAD/issues/8887#issuecomment-1478551688, is the orientation of the object in FreeCAD. To get a usable DXF (or DWG) export from the object whose main face is plane-parallel to the global XZ plane:

1. Check "Use legacy python exporter" in the [preferences](https://wiki.freecad.org/Import_Export_Preferences#DXF).
2. Also check "Project exported objects along current view direction".
3. Call [Std_ViewFront](https://wiki.freecad.org/Std_ViewFront) before exporting.

Roy-043, 2023-08-30:
I am going to go ahead and close this now. Please reopen if you disagree.

closed 2023-08-30 by Roy-043