FreeCAD/FreeCAD #8897: [Problem]

kesslwovv, 2023-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.2.61
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

FreeCad crashes when I change the format specifier to something like `%s` in the format specifier of a line.

This is not something experienced users would do, but to me as a beginner it was not obvious that a (seemingly small) mistake can crash FreeCad.

 Ideally there would have been an error Message like `Invalid Format Specifier` or something in the report view, but most importantly there should be no crash (this lost me the progress of like an hour)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-15:
Firstly could you please fill in a more comprehensive title not just [Problem]

Using Linux Conda build I do get a nice warning `22:37:22  Warning - no numeric format in formatSpec % - Dimension` and no crash so this maybe a MacOS specific issue.

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * Curves 0.6.8
  * PieMenu 1.2.4
  * Plot 2022.4.17
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.54
  * sheetmetal 0.2.59
[/code]

@WandererFan is this a known issue on MacOS versus Linux/Windows?

WandererFan, 2023-03-15:
It breaks in current master.  I haven't tried any other builds yet.

WandererFan, 2023-03-16:
PR #8901 / commit 328cd863a1  fixes this in master branch.

luzpaz, 2023-03-17:
@WandererFan are you planning on backporting ?

WandererFan, 2023-03-17:
PR #8930 fixes this in v0.20

donovaly, 2023-03-17:
Thanks for the fix. It will be available in FreeCAD 0.20.3 (which is planned mid of April)

WandererFan, 2023-03-18:
closed as completed.

commit c20529d5 (2023-03-17): [TD]fix #8897 in v0.20
closed 2023-03-18 by WandererFan