FreeCAD/FreeCAD #8900: [Issue] Mirror doesnt work properly in PD

domovinavi, 2023-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * yaml-workspace 0.1.2
  * 3DfindIT 1.2.0
  * A2plus 0.4.60h
  * fasteners 0.4.54
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

Hello,
the part that i want to mirror doesnt appear. Earlier steps do appear, but if it gets more complex it wont function. 
You can see the attachments in here.
[image]

[image]

Hope you can fix that bug for future versions!
Saludos

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-16:
It would be really useful to have a copy of your file, please zip it in order to attach it here. There was a reported issue with PD Mirroring in the past and it was fixed for that instance https://forum.freecad.org/viewtopic.php?style=10&f=3&t=64900

adrianinsaval, 2023-03-28:
no action is possible if file is not provided, how long should we wait for an answer before closing? This one is likely just misuse, like mirroring only one feature instead of all required to reproduce the shape with transformation.

luzpaz, 2023-03-29:
True. Closing due to inactivity on OP's part

closed 2023-03-29 by luzpaz