FreeCAD/FreeCAD #8902: [Problem] Segfault

xxxajk, 2023-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32311 (Git)
Build type: Unknown
Branch: master
Hash: 44864bf5250193b3405940d57ffcc79f8dd16af4
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * MeshRemodel 1.8919.0
  * fasteners 0.4.23
  * Silk 1.0.0
  * BOLTSFC
  * CurvedShapes 1.0.4
  * Plot 2022.4.17
  * FCGear
  * AirPlaneDesign 0.4.0
  * ThreadProfile 1.84.0
  * Design456 0.0.1
  * parts_library
  * pcb
  * Defeaturing 1.2.0
  * A2plus 0.4.60h
[/code]
```

### Subproject(s) affected?

Part

### Problem description

From Python console or macro.

```
V=App.Vector
c=Part.makeCircle(2000,V(0,0,0))
w=Part.Wire(c)
v=V(-2000,0,0)
w.section(v)

```
:-)

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7ff7926b1090]
#1  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyType_IsSubtype+0x10) [0x7ff793fc4e50]
#2  0x7ff7088d090d in Part::TopoShapePy::section(_object*) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/Mod/Part/Part.so+0x44f
#3  0x7ff7088bef32 in Part::TopoShapePy::staticCallback_section(_object*, _object*) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/Mod/Part/Part.so+0xa6
#4  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8697) [0x7ff794025697]
#5  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyObject_MakeTpCall+0xab) [0x7ff794025b2b]
#6  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74df3) [0x7ff793df1df3]
#7  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7d86) [0x7ff793df9ef6]
#8  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb) [0x7ff793f47e4b]
#9  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x42) [0x7ff793f481d2]
#10  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCode+0x1f) [0x7ff793f485bf]
#11  0x7ff79706a3a7 in Gui::InteractiveInterpreter::runCode(PyCodeObject*) const from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0xfd
#12  0x7ff79706a1d9 in Gui::InteractiveInterpreter::runSource(char const*) const from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0xc3
#13  0x7ff79706a9c9 in Gui::InteractiveInterpreter::push(char const*) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0xc9
#14  0x7ff79706cb94 in Gui::PythonConsole::runSource(QString const&) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0x11a
#15  0x7ff79706c115 in Gui::PythonConsole::keyPressEvent(QKeyEvent*) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0x4a1
#16  0x7ff79388f77d in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x74d
#17  0x7ff79393cd52 in QFrame::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x22
#18  0x7ff79393f9d1 in QAbstractScrollArea::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x251
#19  0x7ff793a15ab9 in QPlainTextEdit::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xc9
#20  0x7ff79384ca66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#21  0x7ff79385681e in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa5e
#22  0x7ff796cfa707 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0x111
#23  0x7ff792e3280a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#24  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc216) [0x7ff7938ae216]
#25  0x7ff79384ca66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#26  0x7ff7938560f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#27  0x7ff796cfa707 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0x111
#28  0x7ff792e3280a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#29  0x7ff793217b2b in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x16b
#30  0x7ff79321d0f1 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x241
#31  0x7ff7931f735b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#32  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x7ff78cace32e]
#33  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7ff790ade17d]
#34  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7ff790ade400]
#35  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7ff790ade4a3]
#36  0x7ff792e8a435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#37  0x7ff792e313ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#38  0x7ff792e39116 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#39  0x7ff796ba14cc in Gui::Application::runApplication() from /home/xxxajk/NetBeansProjects/FreeCAD/freecad-build/lib/libFreeCADGui.so+0x289c
#40  bin/FreeCAD(+0x2dd07) [0x55a3b190fd07]
#41  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7ff792692083]
#42  bin/FreeCAD(+0x2cfae) [0x55a3b190efae]

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xxxajk, 2023-03-16:
Converting to a Vertex seems to fix it, but still :-) Perhaps the conversion could happen for you?

xtemp09, 2023-03-16:
Same on openSUSE 15.4. I see no reason to attach thread backtrace because it's identical to the uploaded, except for Python, which is 3.10.

```
OS: openSUSE Leap 15.4 (KDE//usr/share/xsessions/default)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git) AppImage
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

AjinkyaDahale, 2023-03-17:
The issue is in `src/Mod/Part/App/TopoShapePyImp.cpp:971`

```
            if (PyObject_TypeCheck(item, &(Part::TopoShapePy::Type))) {...}
```

So something is happening that prevents the typecheck itself.

wwmayer, 2023-03-17:
When I run the code snippet I get this expected error message:
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
TypeError: non-shape object in sequence
```

Version is:
```
OS: Ubuntu 22.04.2 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32336 (Git)
Build type: debug
Branch: master
Hash: b1c848b8db1fd816fd2cba48ba062a9318598e9d
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

> So something is happening that prevents the typecheck itself.

Is item a valid pointer?

wwmayer, 2023-03-17:
After doing some more tests I get it crashing after changing the startup wb to Start. In this case the test code must explicitly import Part:
```
import Part
V=App.Vector
c=Part.makeCircle(2000,V(0,0,0))
w=Part.Wire(c)
v=V(-2000,0,0)
w.section(v)
```

AjinkyaDahale, 2023-03-17:
> Is item a valid pointer?

Yes it was non-null and had a non null `ob_type`.

wwmayer, 2023-03-17:
Checking it in a debugger it tells me that pcObj is valid and its value is `Vector (-2000.0, 0.0, 0.0)`. But when checking the elements of the vector it says its ob_refcnt is 0.

AjinkyaDahale, 2023-03-17:
That's curious. I don't know this in details, but I presume `ob_refcnt` should be 1 right?

xtemp09, 2023-03-17:
What does FreeCAD use besides StackWalker to trace crash? What do developers think of [Crashpad](https://github.com/backtrace-labs/crashpad/)?

wwmayer, 2023-03-17:
> What does FreeCAD use besides StackWalker to trace crash? What do developers think of [Crashpad](https://github.com/backtrace-labs/crashpad/)?

Looks interesting and also works on Linux. How do I incorporate it into a project?

> I don't know this in details, but I presume ob_refcnt should be 1 right?

Yes, it should 1. In the meantime I found the root problem. It's hidden in the line `PyObject* item = (*it).ptr();`
The VectorPy class implements the sequence protocol by providing the method sequence_item(), ... Inside there it creates a double object on the fly with `Py_BuildValue("d", a[pos])` and at that point the ref counter is 1. The calling instance is the PyCXX function
```
        virtual T getItem( sequence_index_type i ) const
        {
            return T( asObject( PySequence_GetItem( ptr(), i ) ) );
        }
```
The function asObject() returns a Py::Object and it does NOT increment the counter. If the returned Py::Object is destroyed it decrements the counter to 0 and the double object will be deleted before checking its type.

Since according to the documentation of PySequence_GetItem it returns a new reference the PyCXX code is correct and the function to blame is VectorPy::sequence_item.

So, the problem is caused by creating the double object on the fly instead of holding it inside a list of tuple. This way the ref counter won't be higher than 1 and then it happens that it reaches 0. When it was hold in a list or tuple the function VectorPy::sequence_item would return an object with a ref count of 2 and this way it won't happen that the object will be destroyed to early.

So, the solution is to add a list object to VectorPy and store the double objects there. They will be destroyed together with the VectorPy object.

xtemp09, 2023-03-17:
> Looks interesting and also works on Linux. How do I incorporate it into a project?

Sadly (or fortunately), I don't have experience in implementing a crash-reporting system. I hope these two links will help. [^1] [^2]

The idea of Crashpad is to have another process that monitors the first process activity. In the event of crash it dumps and/or sends the snapshot of the client process.

[^1]: https://github.com/backtrace-labs/crashpad/blob/backtrace/doc/overview_design.md
[^2]: https://support.backtrace.io/hc/en-us/articles/360040516131-Crashpad-Integration-Guide

wwmayer, 2023-03-17:
PR https://github.com/FreeCAD/FreeCAD/pull/8927

closed 2023-03-17 by wwmayer

xxxajk, 2023-03-18:
Nice! Looking forward to the fix pull, which would make the macro (eventual part of a workbench) I'm working on a touch quicker.

AjinkyaDahale, 2023-03-18:
@donovaly this probably needs a backport.

xxxajk, 2023-03-18:
I know this is the wrong place to ask, but, there seems to be no method I can call from python, that takes a list of vectors or vertex to return all  the points (and only the points, not anything else) that intersect (point lands on the wire object exactly). 
The list I need to check is quite large, usually 21,000 to 100,000 points.
At 1ms each vertex, it takes a bit over 21 seconds to check them all, which isn't acceptable.
I use point list in a iteration loop, checking each via wire_object.section(vertex).
This time could be reduced if a list was passed as a collection of points and just have it all done at once in C++, but I have no idea where to even do that, or if it even already exists. If not I'll make a feature request.

AjinkyaDahale, 2023-03-18:
@xxxajk there might already be optimized methods, and probably in OCC itself. Please make a new issue so this can be tracked.

That said 21s doesn't sound particularly bad.

xxxajk, 2023-03-19:
@AjinkyaDahale it is across hundreds of object wires :-) If it was just one set, then I wouldn't complain.
That said, I'll be making two requests that possibly have speedups :-D Also, instead of intersect, I can use a wire object, and isInside but it is actually slower. I have other useful ideas as well, Too many to go into here.

Thank you for letting me know.

AjinkyaDahale, 2023-03-19:
> I have other useful ideas as well, Too many to go into here.

Go ahead and make an issue. Let me know once you've done so.

commit 560752cd (2023-03-17): Base: fixes #8902: Segfault from Part WB functionality in python scripting (#8927)
commit 493213ba (2023-03-20): Base: fix #8902 in v0.20