FreeCAD/FreeCAD #8919: [Problem] Techdraw dimension lines not being recalculated when changing scale

kesslwovv, 2023-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.2.61
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

The dimension lines are not updated properly when changing the scale (in this case 1:5 to 1:4). 

The distances shown in the text are at first the original distance Texts (not updated), when selected and unselected they show their proper distances (they measure the distance between the arrows) but the origin of the line is not anchored to the original vertex.

I found that the following procedure fixes this:
double-clicked the dimension line to open the edit menu
press 'cancel' to exit the menu again

now it is showing the lines properly.

### Anything else?

Wrong (after selecting and unselecting the lines):
[image]

Fixed with above Method:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-03-19:
I'm having trouble duplicating this behaviour.  What scale are you changing?  Please attach your file.
[image]

kesslwovv, 2023-03-19:
I am talking about the user-defined scale option in the projection group.

I made a minimal example (basically a rectangle sketch with a pad):

This is as expected (scale 1:1)
[image]

This is when changing the scale to 1:2
[image]

I re-opened the saved file (looks normal) and then changed the scale back to 1:1
[image]

This is the menu option where I change the scale (in ProjGroup). Note that This is using user-defined scale mode.
[image]

this is the test file:
[test.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11011910/test.FCStd.zip)

WandererFan, 2023-03-20:
OK, I've got it now.  There is a missing recompute somewhere.   

https://wiki.freecad.org/TechDraw_RedrawPage will fix the display.  

Seems to be fixed in v0.21, but I'll take a look to see if it is an easy backport to v0.20.

kesslwovv, 2023-03-20:
thanks. I didn't know this option existed (I was trying the default refresh option but this didn't work)

closed 2023-03-20 by kesslwovv