FreeCAD/FreeCAD #8924: [Problem] FreeCAD will happily overwrite a changed file

sliptonic, 2023-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
any
```

### Subproject(s) affected?

Core

### Problem description

 The user has a document open and has made changes.  
Meanwhile the same document is opened in another instance of FreeCAD, changed, and saved.  
Then the first instance tries to save, FreeCAD happily overwrites the changes.  No warning.

FreeCAD should detect that the file on the disk has changed since it was opened and warn the user before overwriting the file.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-03-22:
OK so looking into this, there's `PropertyString` for the date when the document was last changed, but not an exact time (which should also be an `int` or `float` or some other easy to compare time related format). There still is the modification time stored in the filesystem though.

Zolko-123, 2023-04-17:
> FreeCAD happily overwrites the changes

I've seen this too and I think that what happens in reality is that the original file is renamed to be the backup file, and a new file with the same name is created. So technically the file is not "_overwritten_" but "_replaced_". You can test this very easily:

- create a new file, save-it
- in the OS set permissions to read-only
- modify the file ans save it again

=> an FCStd1 backup file is created with the read-only permissions, and the original file is saved with the same name but read-write permissions

luzpaz, 2025-01-25:
Re-opened as this is a critical problem if it still exists.

luzpaz, 2025-01-25:
Related to #18044

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD/FreeCAD: https://github.com/AstoCAD/FreeCAD/commit/c1e655eecf18e55ef51b05197a27028d334134e9. If someone wants to cherry-pick it and create a PR, these are the relevant commits.