FreeCAD/FreeCAD #8925: [Issue] "Preference packs" table size spreads in height after enabling "Render" workbench

Symbian9, 2023-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git) AppImage
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.6
  * Render 2023.2.2
  * toSketch 1.0.1
  * fasteners 0.4.55
  * sheetmetal 0.2.61
  * Defeaturing 1.2.0
  * dodo 1.0.0
  * Silk 1.0.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * Curves 0.6.8
  * Assembly3 0.11.4
  * freecad.gears 1.0.0
```
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

After enabling "Render" workbench size of _"Preference packs"_ table in _"Preferences > General"_ drastically spreads in height.

After restart FreeCAD it backs to normal size till next switch to "Render" workbench.

### Anything else?

![Screenshot_2023-03-17_17-16-41](https://user-images.githubusercontent.com/13188580/225946301-0d79bdc8-7317-4335-88fa-c52b3dd31fc5.png)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-17:
Therefore would it not be caused by https://github.com/FreeCAD/FreeCAD-render/blob/master/Render/resources/ui/RenderSettings.ui#L9 and therefore raise it with the Render Wb author rather than FreeCAD core?

Symbian9, 2023-03-17:
> therefore raise it with the Render Wb author rather than FreeCAD core?

IDK, but FreeCAD "Preferences" window is affected, not "Render" workbench itself, so it is FreeCAD core issue.

chennes, 2023-03-17:
It's got to be some kind of Qt layout thing. That table is supposed to expand to fill the available space, without forcing the appearance of a scrollbar. I've probably got some setting or other wrong. I wonder if @MisterMakerNL has any ideas, from all the fighting with stylesheets?

MisterMakerNL, 2023-03-18:
I checked it out and it is also screwing with the Techdraw settings as you can see below.
It might be caused by the Qlabel that is named label, which might be a bit to generic.
[image]
And since Qlabel is in like everything it can cause a lot of issues.
Also I don't think but, I am just a newbie in this that it is wise to use super generic name tags. like label_1 and such.
As a newbie I can also definitely understand that it handy to do.
[image]

luzpaz, 2023-03-18:
CC @howetuft

Syres916, 2023-03-18:
Maybe worth adding something along the lines of 'please refrain from using generic names for labels, checkboxes, spinboxes etc' with some good examples to https://wiki.freecad.org/Workbench_creation#Preferences

chennes, 2023-03-19:
It's a little tough since if you are using Qt Designer it defaults to all those generic names. It would mean asking developers to name each and every widget, layout, etc. I grant that it is a good habit, but it's also a bit of busywork. Have we verified that is actually the issue here?

luzpaz, 2023-03-19:
Is this related?  https://github.com/FredsFactory/FreeCAD_AirPlaneDesign/issues/45#issuecomment-1467065889

MisterMakerNL, 2023-03-19:
I did some reading I still think the naming is an issue. But I see there could be other solutions then renaming.
Most Qt projects use a header to define the classes, maybe that can work? Or maybe add some extra "namepace" too the code.

howetuft, 2023-03-19:
Hello,
I've renamed the offending 'label' QLabel. The fix has been pushed to Render master branch. Can someone (@Symbian9?)  confirm it's ok? Many thanks in advance.

luzpaz, 2023-03-19:
Linking https://github.com/FreeCAD/FreeCAD-render/pull/244

MisterMakerNL, 2023-03-19:
I still see the issue after updating the addon :(

luzpaz, 2023-09-29:
What's the diagnosis here ?

luzpaz, 2023-10-11:
bumping for status

luzpaz, 2023-11-08:
@howetuft can you see this issue still ?

howetuft, 2023-12-06:
I looked at it once again, but I can't find anything wrong in my code.
At any rate, my official position is that it is not a Render issue, but a FreeCAD one, although it is triggered by Render. Indeed, I just provide a form to `FreeCADGui.addPreferencePage`, it's up to this method to containerize my form in such a way there is no side-effect on others widgets outside.

MisterMakerNL, 2023-12-06:
Be aware the preference page is completely rebuild, and tho seems to fix the issue at hand!

[image]

luzpaz, 2023-12-24:
Anything left here ?

closed 2024-07-27 by luzpaz