FreeCAD/FreeCAD #8932: [Issue] False positive warning "Selection not restricted to one sketch and its subelements"

PLChris, 2023-03-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.32198 (Git)
Build type: Release
Branch: master
Hash: f51b2156f35399cab38eef1e957a59ad5a11de66
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.54
  * sheetmetal 0.2.61
  * ExplodedAssembly
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

None

### Issue description

When removing an element in Sketcher, I get a warning in report view
`Delete: Selection not restricted to one sketch and its subelements`
This warning clutters report view and doesn't seem to require any action and can hopefully be safely removed.

To reproduce:
- open the attached file
- edit the sketch
- remove the line

-> The aforementioned warning occurs in report view
[warning.FCStd.txt](https://github.com/FreeCAD/FreeCAD/files/11004157/warning.FCStd.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-17:
for some reason I can't download the attached file. I have seen this warning some times when I had selected something in the tree, but also in some unexpected situations. Do you know how to make from scratch a file that has this situation?

luzpaz, 2023-03-18:
[try downloading this warning.FCStd.txt](https://github.com/FreeCAD/FreeCAD/files/11008538/warning.FCStd.txt)

adrianinsaval, 2023-03-18:
downloaded and tested but can't reproduce
```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32311 +2 (Git)
Build type: None
Branch: paddle_button
Hash: 750bc10449cfa0461c81b77baeb6a24b34c041b2
Python 3.10.10, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * fasteners 0.4.55
  * Assembly3 0.11.4
  * CfdOF 1.21.7
  * Plot 2022.4.17
  * Curves 0.6.8
  * Defeaturing 1.2.0
  * DynamicData 2.46.0
  * Dracula 0.0.2
  * 3DfindIT 1.2.0
  * SelectorToolbar
  * FC_SU
  * A2plus 0.4.60k
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * CubeMenu
  * sheetmetal 0.2.61
  * AirPlaneDesign
  * Glass

```

the line is deleted without issues, same on 0.20

PLChris, 2023-03-22:
@looooo, @chennes, can you reproduce on macOS?

PLChris, 2023-05-22:
I retried with the current version after removing ~/Library/FreeCAD and ~/Library/Application Support. Still the same behaviour. The hardware is a MacBook Pro from 2019.
Is there anything I can do to identify the source of the issue?

OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)

adrianinsaval, 2023-05-30:
@abdullahtahiriyo could help out here? What can Chris do to help identify what is going wrong here? The bug seems to be macOS specific

abdullahtahiriyo, 2023-05-31:
@berniev 

Thanks. I do not have macOS and it would be for me trying to chase a ghost.

@PLChris 

At this stage it would be helpful if you could try again. 

If you can still reproduce it, please try to provide much more detailed instructions. 

If it is reproducible, I fear that the difference between reproducing it or not could be in a detail which is not captured in the issue description, or that it is not being correctly interpreted.

Please, let us know.

luzpaz, 2025-01-19:
re-opening since this needs to be verified that i's not an issue anymore

olevski, 2025-01-25:
I just wanted to let you guys know that I got this message. But I wasn't sure if it was real or not. Because I am not that familiar with FreeCAD. But then I was able to make it go away and delete the elements in the sketch (i.e. just some bsplines) by saving the doc, closing freecad and reopening it. Then I could delete the stuff that previously freecad was telling me I could not for the reason that is the title of this  issue.

I am on 1.0.0 using it as an appimage on ubuntu.