FreeCAD/FreeCAD #8935: [Feature Request] Support zero length in Linear Pattern

kerrickstaley, 2023-03-17:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

N/A

### Subproject(s) affected?

PartDesign

### Idea description

Oftentimes with parametric models you can end up with a degenerate Linear Pattern where Occurrences = 1 and Length = 0. When this happens FreeCAD should just create one instance of the object, as if you didn't create a linear pattern. Instead, it currently throws the error "Pattern length too small".

The work-around for now is to add 1 nm (1e-6 mm) to your Length, but it's a little ugly.

The Occurrences = 1 issue was recently fixed, see https://github.com/FreeCAD/FreeCAD/issues/7791 .

### System info
Arch Linux latest
FreeCAD latest Git build (c4b63bc)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-18:
Do you mind opening a forum thread (and bi-bidirectionally linking it) to attract more attention to this FR ?

kerrickstaley, 2023-03-18:
@luzpaz done

luzpaz, 2023-03-18:
@kerrickstaley can you please ask 'Jolbas' what their github handle is so I can assign them this ticket ?

Jolbas, 2023-03-29:
Jolbas is Jolbas here too

Jolbas, 2023-03-31:
The issue can be closed. PR https://github.com/FreeCAD/FreeCAD/pull/9111 is merged.

closed 2023-04-01 by luzpaz