FreeCAD/FreeCAD #8938: On FreeCAD 0.20.2 I can't work on an oblique face

drcowsley, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Problem description

I can't work on an oblique face.  No previous version has given me this problem on this laptop over several years use.

The trivial model demonstrates the problem - making an oblique hole normal to an oblique face.

If I close the report window and try to complete the pocket the whole UI locks up.

[ScaleToolA.zip](https://github.com/FreeCAD/FreeCAD/files/11007019/ScaleToolA.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2023-03-18:
Hmm, I have no explanation. I can reproduce your issue with your file but recreating the sketch fixed the problem for me:
[ScaleToolA-fixed.zip](https://github.com/FreeCAD/FreeCAD/files/11007174/ScaleToolA-fixed.zip)

I cannot reproduce your bug in the way that I can't create a sketch that cannot be used for a hole.
The error message in your file tells me about a wire. But a wire is an object of the Draft workbench. For the Hole feature you must use a sketch (Sketcher workbench).

However, in general, please first open a forum thread before opening a bug report.

closed 2023-03-18 by donovaly

drcowsley, 2023-03-18:
Hi Uwe,

Thanks for replying.  I am a user rather than an expert; I simply use the
Part Design workbench.  I will look in more detail at the different tools,
but it sounds as though I might have gone 'off track' when for the first
time ever I accepted the invitation when installing to make Workbench my
default.

As a non expert I was disappointed to see that I still had to select it
when loading FreeCAD but pleased to see how quickly it appears when I did.

Thank you for your time.  Having drawn the problem to the attention of
somebody who knows more than I do I will back out, leave 'the team' to do
what it does so well (thank you all) and use your advice, and possibly a
reinstall, to get on.

And I will use the forum next time, sorry.

Chris