FreeCAD/FreeCAD #8939: [FEM] timers are moved between non-QThreads

donovaly, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=668245#p668245

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32357 (Git)
Build type: Release
Branch: master
Hash: d23b576c3361b7d593b8e65255a5f970e5954c39
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Installed mods: 
  * Assembly4 0.12.6
  * Curves 0.6.8
  * FEMbyGEN 1.2.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

There are 2 warnings for every iteration of any FEM solver. Here is a screenshot of the problem:
https://forum.freecad.org/viewtopic.php?p=668245#p668245

@abdullahtahiriyo wrote in the forum that the bug is:
"FEM uses multiple threads. Only the main thread, which is a QThread, has an event loop. Between QThreads one can "move" a timer. It is no possible to move a QTimer between non-QThreads. Then any QTimer should then be started and stopped on the main QThread."

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2023-03-18:
Maybe @wwmayer has an idea why this happens because I don't see this in FC 0.20 and my FEM changes were not about Qt threads.

wwmayer, 2023-03-18:
According to the stack trace there are some functions in FEM that are running in the context of another thread and try to perform operations on the GUI.

Unfortunately, from the C++ debugger I cannot see where in Python  the functions are called but they call these functions:
* DocumentPy::removeObject with "Result_mesh" as argument
* FemGui.setActiveAnalysis
* DocumentPy::recompute()

Over several function calls this triggers the TreeWidget to call  statusTimer->start in a non-GUI thread. Although, root of the problem is clearly FEM code it looks like the warning can be avoided if TreeWidget::_updateStatus checks if the current thread is the GUI thread or not.

abdullahtahiriyo, 2023-03-18:
If it it is acceptable for FEM that only one thread controls the UI (and given that without event loop it is not possible to move a QTimer between threads), the option I used for notification is basically to only process the timer if coming from the thread on which the QTimer was started:

https://github.com/FreeCAD/FreeCAD/blob/57f2f06fe22bbac6b74b04413aba991ddbd1016e/src/Gui/NotificationArea.cpp#L827

marioalexis84, 2023-03-19:
With #8951 I still get one (and only one) warning:
`QObject::startTimer: Timers cannot be stated from another thread`

wwmayer, 2023-03-19:
> With https://github.com/FreeCAD/FreeCAD/pull/8951 I still get one (and only one) warning:

What are the steps to reproduce it?

marioalexis84, 2023-03-19:
Is it possible that it is due to the version of Qt?
I did another tests and I can reproduce on Debian stable (Qt 5.15.2) but I can't reproduce on Arch Linux (Qt 5.15.8) using commit f7f78aa44b24dd3ae0bc813c8f67eb230ff933ac.
Now I get two or three warnings `QObject::startTimer: Timers cannot be stated from another thread`. (Last night I only see one).

Steps to reproduce (on Debian):

-Create some mechanical Analysis on Fem workbench using the Elmer solver with elasticity equation or the z88 solver (for example using a Part Box with a constraint fixed on a face and a constraint force on another face).
-In the solver control dialog press Write button.
-Press Run button.
At the end of the analysis, the warnings are printed.

donovaly, 2023-03-20:
For me on Windows I can report that the warnings are now gone. Many thanks.

marioalexis84, 2023-03-20:
> Do you do exactly the same on Debian stable and Arch Linux?

I do the same in both, using commit https://github.com/FreeCAD/FreeCAD/commit/f7f78aa44b24dd3ae0bc813c8f67eb230ff933ac.
- Double click on the solver.
- In the solver control dialog press Write button.
- Press Run button.

File to test: [timer_warning.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/11020647/timer_warning.FCStd.gz)

screenshot from Debian stable: [image]

screenshot from Arch Linux: 
[image]

abdullahtahiriyo, 2023-03-20:
I have compiled Elmer for Ubuntu mate 22.04. I cannot reproduce the QTimer issue.

Is it an option for you to debug in the Debian machine to locate where the warning is being generated?

marioalexis84, 2023-03-21:
>Is it an option for you to debug in the Debian machine to locate where the warning is being generated

I'll try to search in the function calls

marioalexis84, 2023-03-22:
@abdullahtahiriyo @donovaly can you try this?
In the test file that I uploaded above, select the Elmer solver and execute the command `FEM_SolverRun` "Run solver calculations".
This way I get the warnings again in my Arch Linux system, but only with the Elmer solver.

donovaly, 2023-03-22:
> @abdullahtahiriyo @donovaly can you try this? 

I don't get a warning with latest master and on Windows 10.

abdullahtahiriyo, 2023-03-22:
@marioalexis84 

I do not get it either in Ubuntu Mate 22.04.

wwmayer, 2023-03-22:
The used sub-classes of QRunnable do several things on the GUI. At some point it also changes the selection and independent of any solver or data I could reproduce the warning with this snippet:
```
from PySide import QtCore

class Runnable(QtCore.QRunnable):
    def run(self):
        Gui.Selection.clearSelection()

run = Runnable()
QtCore.QThreadPool.globalInstance().start(run)
```

All what have you to do beforehand is to select an object.

wwmayer, 2023-03-22:
https://github.com/FreeCAD/FreeCAD/pull/8987

wwmayer, 2023-03-22:
> Unfortunately I din't find a way to reproduce the crash systematically.

I got a few crashes too and they were related to some rendering issues in Coin3D.

marioalexis84, 2023-03-22:
> I got a few crashes too and they were related to some rendering issues in Coin3D

Yes, I got the same.

I Also got some other crashes in OCCT along with the warning:
`QObject::~QObject: Timers cannot be stopper from another thread`

marioalexis84, 2024-12-30:
The warning `QObject::startTimer: Timers cannot be stated from another thread` comes from a sequence of threads started by the code in femsolver framework.
From one of them the value of some object properties is modified, so the QTimer used in PropertyView raises the warning (in my case, it ends in some random crash due to some undefined behavior in Coin3D nodes and/or reading VTK data).
I managed to fix the error by avoiding some of the generated threads, but a correct solution requires a restructuring of the code in the femsolver directory.

closed 2025-11-03 by chennes