FreeCAD/FreeCAD #8940: [Issue]

donovaly, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=668153#p668153

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32357 (Git)
Build type: Release
Branch: master
Hash: d23b576c3361b7d593b8e65255a5f970e5954c39
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
```

### Subproject(s) affected?

Core

### Issue description

The message notification feature shows messages always below FreeCAD's main window.

Here is a screencast of a selection mistake. The error message appears below the main window:
[image]

### Anything else?

@ab

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2023-03-18:
@abdullahtahiriyo

abdullahtahiriyo, 2023-03-18:
I see.

In my PC this renders inside the frame. I will look into it.

donovaly, 2023-03-20:
Will be fixed with #8958

closed 2023-03-20 by donovaly