FreeCAD/FreeCAD #8943: [Refine shape]  not work when there is "invalid shapes" existing

cidylong, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16131 (Git)
Build type: Release
Branch: (HEAD detached at 0.18.3)
Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

Part

### Issue description

Import a stl file from file system, STL file created by OpenSCAD.
[Create shape from mesh...] in Part ---- ok
select shapes and run [Refine shape], ----- no thing happen.
run [Check Geometry], ---- found there is 1 invalid shapes existing.

My questions is: If there is a invalid shape existing, how run "Refine shape" function to reduce shapes size?

### Anything else?

I tried to include the stl file with out success.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-18:
Can you please discuss on the forum to get more exposure to this issue?

cidylong, 2023-03-28:
@luzpaz, please find attachment.
[corrugated_welle_base_support_end_cut.stl.zip](https://github.com/FreeCAD/FreeCAD/files/11094078/corrugated_welle_base_support_end_cut.stl.zip)

cidylong, 2023-03-28:
Sorry. wrong attachment. please find this one.
[corrugated_welle_swing_inner_end_cut.stl.zip](https://github.com/FreeCAD/FreeCAD/files/11094087/corrugated_welle_swing_inner_end_cut.stl.zip)

cidylong, 2023-03-28:
@luzpaz , please find above attachment. Cidy

adrianinsaval, 2023-04-21:
I don't think this qualifies as a bug, if the shape is invalid and occt does not want to refine there is nothing we can do in the software about it, if you want help to find workarounds for this go to the help forum at forum.freecad.org

chennes, 2023-04-22:
Right, I think this is a help request best answered at the forums, or on Reddit/Facebook/Mastodon/whatever. If it later turns out there's an actionable bug we can open an Issue to work on that.

closed 2023-04-22 by chennes