FreeCAD/FreeCAD #8945: [Issue] Image workbench is suboptimal

PaddleStroke, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=668562#p668562

### Version

0.21 (Development)

### Full version info

```shell
all
```

### Subproject(s) affected?

Core

### Issue description

_Currently :_
Image workbench :
- Image_Open: Open the img in a (bad) image viewer.
- Image_CreateImagePlane: Create a plane and put the selected img in it.
- Image_Scale: a Dialog opens to scale the image.
Core :
- File/Open + File/import : both call Image_Open.
- Double clicking on the Tree element triggers the transform enabling setting the image at correct position.

_Remarks :_
- Image_Open is useless. It only opens the picture in a bad image viewer. It is misleading and bad UI because what the user want is Image_CreateImagePlane.
- Having a workbench for only 1-2 tools is overkill + clutter the workbench selector.

_Suggestion :_
- Delete the Image workbench.
- Move the Image code to Core.
- The image viewer is removed. All tools from the toolbar are removed.
- File/Open + File/import + Paste with an image + drag drop an img, all call Image_CreateImagePlane.
- Double clicking the ImagePlane in tree + in 3DView opens a TaskBox containing the necessary for placement + transform + scale.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-03-18:
> In the Tools menu we have the image creation function and there we can move the function to map an image onto a plane.

@wwmayer you mean `src\Tools\ImageTools\ImageConv` right ? So necessary files should be put in `src\tools\ImageTools\imagePlane` for instance?

But the command CmdCreateImagePlane, it should still be in one of the command tools in src/gui Like CommandStd.cpp?

Currently the Open command connects to the `Py::Object open` of the `ImageGui Module` in a generic way through : `StdCmdOpen` then  `Application::open` which then calls `Command::doCommand(Command::App, "%s.open(u\"%s\")", Module, unicodepath.c_str());`

So I'm unsure how to connect the CmdCreateImagePlane command in here. Should we add an `else if (File.hasExtension("png") || File.hasExtension("jpg") || ...)` in the `Application::open`?

wwmayer, 2023-03-18:
> you mean src\Tools\ImageTools\ImageConv right ?

No. I meant the function in FreeCAD: Tools > Save picture, that creates an image of the currently active view. I thought about adding a function to the Tools menu to also load an image file. Using the global open/import mechanism is an option too but it causes a lot of extra work.

ImageConv has been removed recently and it was a utility to convert between several image formats and had an option to write XPM files to a .cpp file. The utility has become obsolete since we are using Qt's resource framework and switched to the SVG format for most of our icons.

> Should we add an else if (File.hasExtension("png") || File.hasExtension("jpg") || ...) in the Application::open?

No. Like for any other module the core system must register the image file types and define a module name (here FreeCADGui) that is supposed to load them.

PaddleStroke, 2023-03-19:
Thanks for the feedback. And it has already been done I see ! That's great. 
https://github.com/FreeCAD/FreeCAD/pull/8955

PaddleStroke, 2023-03-19:
What do you think about this taskbox to trigger when imagePlane is double clicked : 
[image]

The first rectangle being a 'feature picker'. 
- It displays the name of current orientation plane.
- When clicked it a) opens a menu with 'XY plane' 'XZ plane' & 'YZ plane'. and b) shows the planes on the 3d view
- When user select a feature on the 3d viewer, the edition stops and the newly selected orientation is displayed.

chennes, 2023-03-20:
Can you provide instead an X-scale and a Y-scale with a link that turns on and off uniform scaling (like in Inkscape, for example)?

bambukouk, 2023-03-20:
What seems to be missing as well is setting image transparency.
I use all my images transparent and I believe it can and should be done at the point of importing, please

PLChris, 2023-03-20:
We could then also have a slider or similar for the backlight preference.

bambukouk, 2023-03-20:
Is backlight preference set globally for Display, or is it specific to image only?
(and yes I use both transparency and backlight for my images)

sliptonic, 2023-03-20:
what's the inputbox control for orientation (plane)?  
How would the user align the image to the camera position?  (Yes, it's a border case but might as well include it here)

chennes, 2023-03-20:
> setting position at the point of import is un-important
> I always have to move the image afterwards anyway.

Isn't that what we are talking about here? The interface for working with the image once it's been imported?

bambukouk, 2023-03-20:
> ...How would the user align the image to the camera position? (Yes, it's a border case but might as well include it here)

wouldn't it be better that such a "marginal" case is handled by use of normal placement tool rather than clutter interface for vast majority of users?

adrianinsaval, 2023-03-20:
isn't it better to reuse the dialog and concepts of the attachment dialog rather than making an independent dialog here for positioning the image plane? This will also make it more consistent with other object creation dialogs so the user only needs to learn to use this once.
Then above that we can have a widget with scaling and transparency options.

bambukouk, 2023-03-20:
`> isn't it better to reuse the dialog and concepts of the attachment dialog rather than making an independent dialog here for positioning the image plane? `

in my view no, because the needs are different...
How do you propose to `drag the image on the screen` using placement dialogue?
there needs to be a combination of precise X,Y placement as well as manually dragging on the screen

adrianinsaval, 2023-03-20:
Does the current method allow dragging? You can use the transform tool if you wish to drag things around. In the future we could better unify the attachment, placement and transform dialogs. This would then benefit all objects using that dialog

sliptonic, 2023-03-20:
> > ...How would the user align the image to the camera position? (Yes, it's a border case but might as well include it here)
> 
> wouldn't it be better that such a "marginal" case is handled by use of normal placement tool rather than clutter interface for vast majority of users?

Instead of having three controls (radio buttons) for the three planes, have one combobox with the alignment options (XY XZ, YZ, Camera)  and an apply button.  If other alignment options are desired in the future, they're easily added.

adrianinsaval, 2023-04-21:
Is there something still open here? should we not close this now?

chennes, 2023-04-22:
I agree: if there are still a few lingering problems we need specific Issues to address them, they'll be lost here.

closed 2023-04-22 by chennes