FreeCAD/FreeCAD #8949: [Issue] PartDesign: (extrude) Pad inward taper angle orientation

tbMaker1, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32362 (Git)
Build type: Release
Branch: master
Hash: 45401e30ce29c8a8c66bced6d2db216c96227289
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

There is a difference in taper angle orientation between circles (single edge) and other multi-edge inner sketch items.

For multi-edge inner sketch, the inward taper angle is opposite to the outward taper angle. Or, it can be usefull to be able to have it in the same direction.

### Anything else?

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-20:
isn't this a duplicate issue?

tbMaker1, 2023-03-20:
Yes and no. It was not only about the other bug (which is more than just single edge issue), it was about the possibility to adjust the orientation as needed.

adrianinsaval, 2023-03-20:
In that case this should have been made as exclusively about adjusting orientation

tbMaker1, 2023-03-20:
But adjusting orientation per internal feature would not have been the same topic either (because much more complex and probably not adequate).
Here, the simplification assumed was that all internal feature would be in one way or the other. 
Therefore the implication that the discrepency could/would be removed.

adrianinsaval, 2023-03-20:
sorry if I wasn't clear but that is not what I meant, what I mean is that we should leave the problem with circles in it's own issue as it is not related to this one. Make it exclusively about allowing to choose to have parallel or opposite orientation.

tbMaker1, 2023-03-20:
Ok

closed 2023-03-21 by tbMaker1