FreeCAD/FreeCAD #8954: [Issue] STL Export produces invalid result

topse, 2023-03-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60k
  * sheetmetal 0.2.59
[/code]
```

### Subproject(s) affected?

None

### Issue description

I am using FreeCAD for 3D printing. The current model I an working on seem to get invalid when exporting in STEP, OBJ or STL (I did not try other formats). Screenshots from slicer:

1. Before Slicing 
[image]

2. After Slicing
[image]

You can see the slicer leaves out a big area. As I am using Prusa Slicer 2.5.1 there is a function "Fix through netfabb", but if I use that, the drill hole gets "repaired away" :-)

I attached the project file below.

### Anything else?

[MovingJaw.zip](https://github.com/FreeCAD/FreeCAD/files/11009487/MovingJaw.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-18:
Latest stable is 0.20.2 (you're using 0.20.1)
Please test on latest stable and report back.

Also, If you have the ability, please test development version 0.21.x https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

topse, 2023-03-18:
Sorry -- just found current version and updated the report (after testing). 

Same with this development version:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32258 (Git)
Build type: Release
Branch: master
Hash: 0f4acc55beb66dcc9dfab442344c6286d6a2913e
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60k
  * sheetmetal 0.2.59
[/code]

Syres916, 2023-03-18:
This isn't FC version specific, if the user looks inside the hole with the thread they can see that the Boolean has been setup correctly (it's cut the surface open), this is entirely user error. If a forum discussion had been created then there would be more 'eyes'. I personally wouldn't have even started modelling this way but that's personal preference.

luzpaz, 2023-03-18:
Thanks @Syres916  
Closing

closed 2023-03-18 by luzpaz

topse, 2023-03-18:
I am very sorry... next time I get to the forum first.

Thanks and Best Regards,
Tobias

Syres916, 2023-03-18:
Attached are two ways to model it, one using Part Wb and Sketcher Wb and the other using Part Design 
[MovingJaw_Syres.zip](https://github.com/FreeCAD/FreeCAD/files/11009590/MovingJaw_Syres.zip)