FreeCAD/FreeCAD #8965: Sketch tools: Pad fails with wire not closed in version 0.20.1 while Pad on the same file in version 0.19 works

bmateev, 2023-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * fasteners 0.4.55
  * Glass
  * Help 1.0.3
  * OpticsWorkbench 1.0.9
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Render 2023.2.2
  * Rocket 2.8.6
  * Silk 1.0.0
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
  * workfeature
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

A simple sketch cannot be padded. The sketch is fully constrained and wire is closed.
Same project opened in version 0.19 has no issues and padding works. 

After padding in 0.19, opening the project in 0.20.1 shows the pad properly without errors. If tried to modify the pad spits error "Wire not closed" and marks the pad as having an issue.

Attached is the project. Try modifying the sketch to replicate the issue.

[image]
[shower redirect.zip](https://github.com/FreeCAD/FreeCAD/files/11014134/shower.redirect.zip)

### Anything else?

Attached the project file in the issue section. (shower redirect.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bmateev, 2023-03-21:
I see now the extra arc. The arc (it was supposed to be only one) was created from a circle with "trim edge" tool. Not sure why 2 arcs were created. Should I assume v0.20.1 is unstable and update to v0.20.2?

Thank you for the prompt replies! You guys are great!

Syres916, 2023-03-21:
I wouldn't say 0.20.1 is unstable but if you look at https://github.com/FreeCAD/FreeCAD/releases/tag/0.20.2 to see how bug fixes have been applied it's worth the upgrade. Hopefully in a couple of months I would hope 0.20.3 is ready with even more polish. If you're happy to close this issue then please feel free to complete, thanks.

luzpaz, 2023-03-21:
Closing ticket

closed 2023-03-21 by luzpaz