FreeCAD/FreeCAD #8968: [Problem] C++ DXF importer does not properly import long (>250 characters) text

kpemartin, 2023-03-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

DXF file is for AutoCAD version 13.
```

### Subproject(s) affected?

None

### Problem description

If an AutoCAD drawing contains an MTEXT entity whose text exceeds 250 characters, the corresponding DXF file content has the text broken up into multiple records. There is a series of type 3 records each containing 250 characters of text, followed by a single type 1 record that contains the remaining part of the text.

Here is what it looks like in AutoCAD:
[image]
Note that I have drawn lines around the text to show where it belongs

The import options should include the "Import texts and dimensions" checkbox.

The C++ importer does not process the type 3 records, so you only get the tail end of the text; the length of this tail is about (original length mod 250).
[image]

The C++ importer also places the text improperly (it should be below the horizontal line) but that is not the topic of this Issue.

The legacy importer imports all the text and places it properly.
[image]

Neither importer wraps the text within the width specified in the DXF file, but that is not the topic of this Issue.

Note that due to #8884 (where the C++ importer does not import text at all) this problem cannot be currently observed in the development version.

The attached zip file contains the DXF file being imported.

I have a PR in the works with a fix.

### Anything else?

[LongText.zip](https://github.com/FreeCAD/FreeCAD/files/11018896/LongText.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-03-22:
Should this be closed, @kpemartin ? Or do you want to work any further on wrapping?
(Note FYI: If you use "Closes 8968" in your commit messages it autocloses the related issue)

kpemartin, 2023-03-23:
Yorik, I think I'll stay away from the auto-closing until I'm more familiar with the github workflow.

I've noticed that on the right sidebar for the issue, under Development, it shows "No branches or pull requests". I would expect this to mention PR #8973 and I was wondering if this was normal to be blank.

luzpaz, 2023-03-23:
> I've noticed that on the right sidebar for the issue, under Development, it shows "No branches or pull requests"

@kpemartin they are the same https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

kpemartin, 2023-06-11:
One of my other changes, #9060, incorrectly removed one of the fixes from #8973 for this issue, so this issue is live again.