FreeCAD/FreeCAD #8970: [Feature Request] Two feature ideas

xxxajk, 2023-03-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76934

### Subproject(s) affected?

Other (specify in description)

### Idea description

Totally lacking is the ability to check multiple points in one go.
This should be something that is trivial to even make happen in parallel, because we can cache the list results before touching anything in Python's area. We are only doing a simple read of a large list, and returning another one that is just a shallow copy (pointers).

As stated in the forum...
Here's the current problem:
I have to wait 6 hours for something that should be able to happen in under a minute, especially on a current modern 64bit machine.
And yeah, I'm aware of the Python GIL, I dislike it too. :-) Doesn't mean that OCC couldn't leverage threads and parallelize...

Why so long? Because I have to iterate each point, and there's thousands of them, and each vector check eats 1ms. After 1k points you are already at 1 second, not including the Python overhead, which isn't much.
3,600,000 points takes an hour to calculate if it's "inside" or not, however an object with this many points displays pretty much instantly -- if this happened with something like an STL, well, I think you get the idea.

First one... 
```
pts =list of vectors, or vertexes
w = Part.Wire(some_edges) # could be a wire or face here...
w.areInside([list of vectors or vertexes], tolerance, checkFace) 
```

This is like `isInside()` but for a list of points, and returns a list of points that all are inside, instead of a boolean. Of course it could return a list of booleans that match too. either way I'd be happy.

Second idea is the same above but for a list of `Part.LineSegment()` This would almost be like doing section() except allowing just lines, not a wire.  the C++ could Take start/end points, and use the previous idea (all in parallel). Rule would be the entire line (both points) must pass `isInside()` in order to be true. As an addional idea, a third possibility is also possible, where if only one point is inside, a None would be able to handle that case. Wouldn't tell you which point, but it doesn't matter in my use case.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xxxajk, 2023-03-20:
They're the same thing, tbh :-) Both find points, just the type is different.
I'll be preparing a demo and post it on the forum and here later tonight. I should be able to whip up a demonstration that shows what I mean.

xxxajk, 2023-03-21:
Here's the demo...
[test.FCMacro.txt](https://github.com/FreeCAD/FreeCAD/files/11024382/test.FCMacro.txt)

xtemp09, 2023-03-21:
@xxxajk, have you tried to parallelize the macro you posted? I don't really know Python, but it looks like a sequential script.

And I agree, parallelization using SYCL, OpenCL or even over CPU cores will be beneficial.

luzpaz, 2023-03-21:
Please update ticket name with less generic title, TIA

xxxajk, 2023-04-06:
Turns out (and you might find this interesting) that if you unblock the GIL and restore it in the right places, these C++ methods can thread. They're still awfully slow though, but that's OCC's fault, especially since everything I am testing is "technically" 2d. OCC should realize this as a fact, and do the correct thing.
Meanwhile, I've resorted to something a little bit better, that avoids OCC as much as I can.