FreeCAD/FreeCAD #8976: Arc tool with end points and point along the arc modifies the sketch. Need Ctrl-Z to perform as expected. [Latest v0.20.2]

bmateev, 2023-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * fasteners 0.4.55
  * Glass
  * Help 1.0.3
  * OpticsWorkbench 1.0.9
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Render 2023.2.2
  * Rocket 2.8.6
  * Silk 1.0.0
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
  * workfeature
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

When using the arc tool with 2 end points and a point along the arc errors are thrown and sketch is modified completely. After Ctrl-Z is pressed the expected resulting sketch is presented. Seems there are underlying issues with handling the calculations of this operation. Attached are 2 videos in order for the issues to be recreated. No need for a freecad project as I isolated the particular issue with the arc tool. 

[Arc fails.webm](https://user-images.githubusercontent.com/22362713/226508878-bf159a4b-9bcd-49b0-9014-75a27109b43d.webm)

https://user-images.githubusercontent.com/22362713/226508901-594e87fe-243b-4eb7-9f23-57f091c59e0e.mp4

### Anything else?

Videos are attached in Issue description section.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-03-29:
Could you attach a file for this?

luzpaz, 2023-06-10:
@bmateev please attach example file (zip it and drag drop in to github text field). TIA

maxwxyz, 2024-10-30:
Closing this for no further feedback.

closed 2024-10-30 by maxwxyz