FreeCAD/FreeCAD #8978: [Issue] Sketch vertices disappear in edit mode

AjinkyaDahale, 2023-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

N/A

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32412 +14 (Git)
Build type: Debug
Branch: sketcher-draw-with-knots (only available build right now, also seen without these changes)
Hash: 23787d707f044c8c31b3580612fdcefaaf991d55
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/India (en_IN)
```

### Subproject(s) affected?

Sketcher

### Issue description

To replicate:

1. Create two lines with one common point (say with polygon tool). This may be possible with other cases. A case where this bug exists is attached.
2. Try to drag a line around.

Observed: vertices of the lines may disappear. This doesn't always happen and sometimes the vertices reappear.

[example-disappearing-verts.zip](https://github.com/FreeCAD/FreeCAD/files/11032169/example-disappearing-verts.zip)

https://user-images.githubusercontent.com/4316933/226701175-1ff60f78-d908-4594-b5c0-eed79c465833.mp4

### Anything else?

First reported here: https://github.com/FreeCAD/FreeCAD/pull/8530#issuecomment-1474933279.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-08:
@PaddleStroke is this on your radar ?

PaddleStroke, 2023-11-09:
No. And I can't replicate this problem.

AjinkyaDahale, 2023-11-09:
> No. And I can't replicate this problem.

Can't replicate either with the following.

```
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) Snap 828
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

CAN replicate with the following:
```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 +7 (Git)
Build type: RelWithDebugFlags
Branch: core-access-doc-properties-from-expressions
Hash: eb1e6250e63d4f6c616d274ec172806f760f8258
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * PreVes
  * Curves 0.5.12
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

luzpaz, 2023-11-10:
How can a theme affect sketcher performance...weird

AjinkyaDahale, 2023-11-10:
> How can a theme affect sketcher performance...weird

Well until someone can reproduce the bug again, I doubt we'll know. If only had the foresight to create a preference pack of my current setting before changing the theme.

AjinkyaDahale, 2023-11-10:
Closing it as "cannot reproduce".

closed 2023-11-10 by AjinkyaDahale