FreeCAD/FreeCAD #8979: SIGSEGV Segmentation after opening a second dialog.

paulvdhoeven, 2023-03-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git) AppImage
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Installed mods: 
  * freecad.gears 1.0.0
  * fasteners 0.4.55
  * persistenttoolbars
  * kicadStepUpMod 10.17.1
  * Glass (Disabled)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Sometimes I push buttons in an inattentive way and FreeCAD sometimes crashes while doing so. In an attempt to pin it down I found a reliable way to do this.

First draw a simple object as preparation:
* New File / New body / New Sketch / draw rectangle / Close sketch / Pad the sketch

This works as expected, but then:
1). Double click the pad in the tree, this opens a "Set color per face" dialog in the combo view.
2). Pretend to forget you opened that.
3). Double click on the Sketch  in an attempt to edit the sketch. FreeCAD asks:
[image]

4). Click on the  **YES** button, and FreeCAD crashes immediately.

I have repeated this at least 5 times in a row while creating this bug report, but I've seen it happen some years ago in older versions too.

If FreeCAD was started from the command line, it spits out some text, which I have copy / pasted below.

[code]
paul@cezanne:~$ freecad
FreeCAD 0.21.0, Libs: 0.21.0R31917 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

NOTICE: Addon "/home/paul/.local/share/FreeCAD/Mod/Glass" disabled by presence of ADDON_DISABLED stopfile
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
NOTICE: Addon "/home/paul/.local/share/FreeCAD/Mod/Glass" disabled by presence of ADDON_DISABLED stopfile
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7fc732ae9090]
#1  0x7fc734c35084 in Gui::Document::getMDIViews[abi:cxx11]() const from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x14
#2  0x7fc734c53d20 in Gui::Document::getActiveView() const from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x40
#3  0x7fc660a356e4 in SketcherGui::ViewProviderSketch::updateData(App::Property const*) from /tmp/.mount_freecaYk0dOF/usr/lib/SketcherGui.so+0x184
#4  0x7fc735074b16 in Gui::ViewProviderDocumentObject::update(App::Property const*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x56
#5  0x7fc734c5260e in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x4e
#6  /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADApp.so(+0x1eac56) [0x7fc7340afc56]
#7  0x7fc7340e5593 in App::DocumentObject::onChanged(App::Property const*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADApp.so+0xd3
#8  0x7fc7341f5365 in App::Property::hasSetValue() from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADApp.so+0x135
#9  0x7fc6c36e2149 in Sketcher::SketchObject::acceptGeometry() from /tmp/.mount_freecaYk0dOF/usr/lib/Sketcher.so+0x39
#10  0x7fc6c36ebcbd in Sketcher::SketchObject::onChanged(App::Property const*) from /tmp/.mount_freecaYk0dOF/usr/lib/Sketcher.so+0x2dd
#11  0x7fc7341f5365 in App::Property::hasSetValue() from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADApp.so+0x135
#12  0x7fc6c8690aac in Part::PropertyGeometryList::setValues(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> >&&) from /tmp/.mount_freecaYk0dOF/usr/lib/Part.so+0x21c
#13  0x7fc6c8690ca1 in Part::PropertyGeometryList::setValues(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> > const&) from /tmp/.mount_freecaYk0dOF/usr/lib/Part.so+0xc1
#14  0x7fc6c36c4619 in Sketcher::SketchObject::solve(bool) from /tmp/.mount_freecaYk0dOF/usr/lib/Sketcher.so+0x279
#15  0x7fc660a36075 in SketcherGui::ViewProviderSketch::setEdit(int) from /tmp/.mount_freecaYk0dOF/usr/lib/SketcherGui.so+0x7c5
#16  0x7fc73505ae44 in Gui::ViewProvider::startEditing(int) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x14
#17  0x7fc73507ada5 in Gui::ViewProviderDragger::startEditing(int) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x15
#18  0x7fc734c56815 in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0xb45
#19  0x7fc660a17584 in SketcherGui::ViewProviderSketch::doubleClicked() from /tmp/.mount_freecaYk0dOF/usr/lib/SketcherGui.so+0x24
#20  0x7fc734ec4865 in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x385
#21  0x7fc7338fc1ca in QWidget::event(QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5+0x148
#22  0x7fc733966791 in QFrame::event(QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5+0x27
#23  0x7fc7330427a6 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Core.so.5+0xe6
#24  0x7fc7338d2060 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5+0xc8
#25  0x7fc7338d6357 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5+0x663
#26  0x7fc734c89b09 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x89
#27  0x7fc73304293e in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Core.so.5+0x8e
#28  0x7fc7338d596b in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5+0x283
#29  /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5(+0x1a3345) [0x7fc73390b345]
#30  /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5(+0x1a44bd) [0x7fc73390c4bd]
#31  0x7fc7338d2071 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#32  0x7fc734c89b09 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x89
#33  0x7fc73304293e in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Core.so.5+0x8e
#34  0x7fc73340e65b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Gui.so.5+0x9f3
#35  0x7fc7333fae39 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Gui.so.5+0x57
#36  /tmp/.mount_freecaYk0dOF/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5658e) [0x7fc72d8ea58e]
#37  /tmp/.mount_freecaYk0dOF/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x25c) [0x7fc72f4bd1bc]
#38  /tmp/.mount_freecaYk0dOF/usr/bin/../lib/./libglib-2.0.so.0(+0x58489) [0x7fc72f4bd489]
#39  /tmp/.mount_freecaYk0dOF/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7fc72f4bd551]
#40  0x7fc733073a90 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Core.so.5+0x5c
#41  0x7fc7330404fe in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Core.so.5+0x180
#42  0x7fc7330446d9 in QCoreApplication::exec() from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libQt5Core.so.5+0x107
#43  0x7fc734c0692e in Gui::Application::runApplication() from /tmp/.mount_freecaYk0dOF/usr/bin/../lib/libFreeCADGui.so+0x1aee
#44  /tmp/.mount_freecaYk0dOF/usr/bin/freecad(+0x68e5) [0x5595202608e5]
#45  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fc732aca083]
#46  /tmp/.mount_freecaYk0dOF/usr/bin/freecad(+0x6c29) [0x559520260c29]

[/code]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-21:
Are you running the latest dev version ?

paulvdhoeven, 2023-03-22:
No, but close.
I already attached the version info to my opening post, It is a weekly build:
Version: 0.21.0.31917 (Git) AppImage
with date 2023-03-05

I had also seen this same problem for over a year ago, so it seems unlikely something significant changed in the last two weeks.

I also find it curious that a double click on the pad opens this **Set color per face** dialog in the combo view:
[image]
My expectation would be for it to open the **Edit Pad** dialog, and that is why I tend to get into the next step where FreeCAD crashes.

xtemp09, 2023-03-22:
Same on Windows.

```c++
* thread #1, stop reason = Exception 0xc0000005 encountered at address 0x7ffcf169dd1b: Access violation reading location 0x000001d0
    frame #0: 0x00007ffcf169dd1b FreeCADGui.dll`class std::list<class Gui::MDIView *, class std::allocator<class Gui::MDIView *>> Gui::Document::getMDIViews(void) const + 75
FreeCADGui.dll`class std::list<class Gui::MDIView *, class std::allocator<class Gui::MDIView *>> Gui::Document::getMDIViews(void) const:
->  0x7ffcf169dd1b <+75>: movq   0x1d0(%r14), %rax
    0x7ffcf169dd22 <+82>: movq   0x130(%rax), %rax
    0x7ffcf169dd29 <+89>: movq   (%rax), %rbx
    0x7ffcf169dd2c <+92>: cmpq   %rax, %rbx
```

<details><summary>Full backtrace</summary>

#### Backtrace obtained by _lldb_ version 15.0.7

```c++
  * frame #0: 0x00007ffcf169dd1b FreeCADGui.dll`class std::list<class Gui::MDIView *, class std::allocator<class Gui::MDIView *>> Gui::Document::getMDIViews(void) const + 75
    frame #1: 0x00007ffcf169d78d FreeCADGui.dll`class Gui::MDIView * Gui::Document::getActiveView(void) const + 45
    frame #2: 0x00007ffcbfffeb18 SketcherGui.pyd`void SketcherGui::ViewProviderSketch::updateData(class App::Property const *) + 392
    frame #3: 0x00007ffcf196a434 FreeCADGui.dll`void Gui::ViewProvider::update(class App::Property const *) + 132
    frame #4: 0x00007ffcf19741e6 FreeCADGui.dll`void Gui::ViewProviderDocumentObject::update(class App::Property const *) + 182
    frame #5: 0x00007ffcf16a7edd FreeCADGui.dll`void Gui::Document::slotChangedObject(class App::DocumentObject const &, class App::Property const &) + 189
    frame #6: 0x00007ffcf25b377b FreeCADApp.dll`void * App::Document::create(void) + 939
    frame #7: 0x00007ffcf2575e35 FreeCADApp.dll`bool App::DocumentT::operator==(class App::DocumentT const &) const + 853
    frame #8: 0x00007ffcf259d2ae FreeCADApp.dll`bool App::CellAddress::operator>(struct App::CellAddress const &) const + 13054
    frame #9: 0x00007ffcf25d5054 FreeCADApp.dll`void App::DocumentObject::onChanged(class App::Property const *) + 1812
    frame #10: 0x00007ffcc3873b1a Part.pyd`void Part::Feature::onChanged(class App::Property const *) + 250
    frame #11: 0x00007ffcc0ec9a0a Sketcher.pyd`void Sketcher::SketchObject::onChanged(class App::Property const *) + 1386
    frame #12: 0x00007ffcf2672f22 FreeCADApp.dll`void App::Property::hasSetValue(void) + 50
    frame #13: 0x00007ffcc3889792 Part.pyd`void Part::PropertyGeometryList::setValues(class std::vector<class Part::Geometry *, class std::allocator<class Part::Geometry *>> &&) + 722
    frame #14: 0x00007ffcc38898d1 Part.pyd`void Part::PropertyGeometryList::setValues(class std::vector<class Part::Geometry *, class std::allocator<class Part::Geometry *>> const &) + 193
    frame #15: 0x00007ffcc0ed2243 Sketcher.pyd`int Sketcher::SketchObject::solve(bool) + 723
    frame #16: 0x00007ffcbfffd2c2 SketcherGui.pyd`bool SketcherGui::ViewProviderSketch::setEdit(int) + 2594
    frame #17: 0x00007ffcf196a358 FreeCADGui.dll`class Gui::ViewProvider * Gui::ViewProvider::startEditing(int) + 24
    frame #18: 0x00007ffcf1977299 FreeCADGui.dll`class Gui::ViewProvider * Gui::ViewProviderDragger::startEditing(int) + 25
    frame #19: 0x00007ffcf16a6f44 FreeCADGui.dll`bool Gui::Document::setEdit(class Gui::ViewProvider *, int, char const *) + 4996
    frame #20: 0x00007ffcbffef19b SketcherGui.pyd`bool SketcherGui::ViewProviderSketch::doubleClicked(void) + 43
    frame #21: 0x00007ffcf1842cd9 FreeCADGui.dll`class Gui::MenuItem & Gui::MenuItem::operator=(class Gui::MenuItem const &) + 78905
    frame #22: 0x00007ffcf04179c8 Qt5Widgets.dll`bool QWidget::event(class QEvent *) + 328
    frame #23: 0x00007ffcf04b8ee0 Qt5Widgets.dll`bool QFrame::event(class QEvent *) + 48
    frame #24: 0x00007ffcf0618378 Qt5Widgets.dll`bool QAbstractItemView::viewportEvent(class QEvent *) + 776
    frame #25: 0x00007ffcc7a94b5a Qt5Core.dll`bool QCoreApplicationPrivate::sendThroughObjectEventFilters(class QObject *, class QEvent *) + 218
    frame #26: 0x00007ffcf03f497c Qt5Widgets.dll`bool QApplicationPrivate::notify_helper(class QObject *, class QEvent *) + 252
    frame #27: 0x00007ffcf03f28b0 Qt5Widgets.dll`bool QApplication::notify(class QObject *, class QEvent *) + 1872
    frame #28: 0x00007ffcf16c4ea9 FreeCADGui.dll`void Gui::ExpressionTextEdit::textChanged2(class QString, int) + 3385
    frame #29: 0x00007ffcc7a92aca Qt5Core.dll`bool QCoreApplication::notifyInternal2(class QObject *, class QEvent *) + 186
    frame #30: 0x00007ffcf03f5b95 Qt5Widgets.dll`bool QApplicationPrivate::sendMouseEvent(class QWidget *, class QMouseEvent *, class QWidget *, class QWidget *, class QWidget **, class QPointer<class QWidget> &, bool, bool) + 965
    frame #31: 0x00007ffcf044040a Qt5Widgets.dll`QSizePolicy::QSizePolicy(void) + 11578
    frame #32: 0x00007ffcf043e38e Qt5Widgets.dll`QSizePolicy::QSizePolicy(void) + 3262
    frame #33: 0x00007ffcf03f4990 Qt5Widgets.dll`bool QApplicationPrivate::notify_helper(class QObject *, class QEvent *) + 272
    frame #34: 0x00007ffcf03f3a13 Qt5Widgets.dll`bool QApplication::notify(class QObject *, class QEvent *) + 6323
    frame #35: 0x00007ffcf16c4ea9 FreeCADGui.dll`void Gui::ExpressionTextEdit::textChanged2(class QString, int) + 3385
    frame #36: 0x00007ffcc7a92aca Qt5Core.dll`bool QCoreApplication::notifyInternal2(class QObject *, class QEvent *) + 186
[... cut ...]
    frame #46: 0x00007ff6f1374aa6 FreeCAD.exe
    frame #47: 0x00007ffd2c447034 kernel32.dll`BaseThreadInitThunk + 20
    frame #48: 0x00007ffd2cee2651 ntdll.dll`RtlUserThreadStart + 33
```

</details>

<details><summary>About</summary>

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: master
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
</details>

wwmayer, 2023-03-24:
> I also find it curious that a double click on the pad opens this Set color per face dialog in the combo view:

Weird! For me it opens the Pad dialog as expected. When I want to change the color per face dialog then it only works via context-menu and pressing the second entry there.
To figure out what's happening you can set a break point here https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/PartDesign/Gui/ViewProvider.cpp#L63 and here https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/PartDesign/Gui/ViewProvider.cpp#L85 and check the value of ModNum.

abdullahtahiriyo, 2023-03-24:
@wwmayer 

I am at it.

wwmayer, 2023-03-24:
I found a way to open the color dialog when double-clicking: under Edit > Edit mode you can select the mode that should be used when double-clicking an object. If you set it e.g. to Color then double-clicking the pad indeed opens the color dialog. But this is correct and intended behaviour, then.

abdullahtahiriyo, 2023-03-24:
@paulvdhoeven 

Thank you very much for creating this issue and provided those simple steps to reproduce the issue.

paulvdhoeven, 2023-03-24:
> I found a way to open the color dialog when double-clicking: under Edit > Edit mode ...

I can confirm that **Edit / Edit Mode** was set to "color" for me, and I have now set it to "default" and FreeCAD does not crash anymore. KiCad only crashes with the "Edit / Edit Mode / Color". The other three ("Default", "Transform", "Cutting") do not Crash FreeCAD.

I see xtemp09 already made a commit / pull request.
At what moment does this issue get closed? Am I supposed to do that, or do one of the developers do it when the fix is merged and/or tested?

PR #9025 (2023-03-24): [Sketcher] Fix segmentation fault in ViewProviderSketch::setEdit
If no one steps up to take on the challenge, then **xtemp09** will. This pull request closes #8979. The issue reported crash using _PartDesign_.

Turns out some dialogs use `Gui::Control().reject()` to close opened dialog while all other dialogs use `Gui::Control().closeDialog()`. Replacement the former with the latter prevents the segmentation fault caused by _Sketcher_. The crash does not occur with other workbenches.

---

**Note 1**. The crash occurs on all platforms, while the issue was reported on Linux Mint 20.3. I checked [... cut]
commit a65a7f67 (2023-03-24): [Sketcher] Fix segmentation fault in ViewProviderSketch::setEdit

Closes #8979
closed 2023-03-24 by wwmayer