FreeCAD/FreeCAD #9000: [Problem] FreeCad crash when trying to fillet an edge

Anthony-Gaudino, 2023-03-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.3.0
  * Render 2023.2.2
  * Silk 1.0.0
  * Defeaturing 1.2.0
  * A2plus 0.4.60k
  * Manipulator 1.5.0
  * sheetmetal 0.2.61
  * toSketch 1.0.1
  * Assembly4 0.12.6
  * Assembly3 0.11.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

There's one specific edge that when selected and then upon clicking on create fillet in Part Design causes FreeCad to crash.
The crash always happens.

I'm creating this issue here, but I think the culprit is OCC.
Bellow is the crash report and minimal file.
[crash_report.txt](https://github.com/FreeCAD/FreeCAD/files/11044808/crash_report.txt)
[crash_part.zip](https://github.com/FreeCAD/FreeCAD/files/11044987/crash_part.zip)

Here's the edge that causes the issue:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-03-24:
Yes, it seems like an OCCT bug. The crash occurs [here](https://github.com/FreeCAD/FreeCAD/blob/c726b69d7ede5b1af82407c515e14375f3358be3/src/Mod/PartDesign/App/FeatureFillet.cpp#L110-L112):

```c++
-> 110     mkFillet.Build();
   111     if (!mkFillet.IsDone())
   112         return new App::DocumentObjectExecReturn("Failed to create fillet");
```

grubertechnic, 2023-07-15:
stacktrace

```
Err: Program received signal SIGSEGV, Segmentation fault.
Err: #0  /lib/x86_64-linux-gnu/libc.so.6(+0x38d60) [0x7f6b7ea3fd60]
Err: #1  0x7f6a472a17f2 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /usr/local/lib/libTKFillet.so.7.7+0x1892
Err: #2  0x7f6a472a5025 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /usr/local/lib/libTKFillet.so.7.7+0x65
Err: #3  0x7f6a4726cf05 in ChFi3d_Builder::Compute() from /usr/local/lib/libTKFillet.so.7.7+0x415
Err: #4  0x7f6a473eee0d in BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&) from /usr/local/lib/libTKFillet.so.7.7+0x1d
Err: #5  0x7f6a43d772d7 in PartDesign::Fillet::execute() from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/_PartDesign.so+0x757
Err: #6  0x7f6b80b4217a in App::DocumentObject::recompute() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0xba
Err: #7  0x7f6a48426369 in Part::Feature::recompute() from /home/user/Programs/FreeCADmaster/build/Mod/Part/Part.so+0x9
Err: #8  0x7f6b80aef3ed in App::Document::_recomputeFeature(App::DocumentObject*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x29d
Err: #9  0x7f6b80af68ae in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x56e
Err: #10  0x7f6b80ba3fa5 in App::DocumentPy::recompute(_object*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x2d5
Err: #11  0x7f6b80ba4154 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x24
Err: #12  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(+0x10cc74) [0x7f6b801a0c74]
Err: #13  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x90) [0x7f6b8015aa50]
Err: #14  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x8632) [0x7f6b8010c2f2]
Err: #15  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(+0x1a4ddc) [0x7f6b80238ddc]
Err: #16  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x4e) [0x7f6b8023912e]
Err: #17  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3e) [0x7f6b8023917e]
Err: #18  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(PyEval_EvalCode+0x1b) [0x7f6b8023472b]
Err: #19  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(+0x1e9735) [0x7f6b8027d735]
Err: #20  /lib/x86_64-linux-gnu/libpython3.9.so.1.0(PyRun_StringFlags+0x93) [0x7f6b8027e013]
Err: #21  0x7f6b80782718 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADBase.so+0x58
Err: #22  0x7f6b8149b7ef in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x7f
Err: #23  0x7f6b8149b94f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0xcf
Err: #24  0x7f6b8149ba53 in Gui::Command::updateActive() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x33
Err: #25  0x7f6a4371a285 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x2c5
Err: #26  0x7f6a4371bd32 in finishDressupFeature(Gui::Command const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Part::Feature*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x6c2
Err: #27  0x7f6a4372879d in makeChamferOrFillet(Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0xbd
Err: #28  0x7f6a4372896c in CmdPartDesignFillet::activated(int) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x3c
Err: #29  0x7f6b8149f42b in Gui::Command::_invoke(int, bool) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x1fb
Err: #30  0x7f6b8149f7de in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0xfe
Err: #31  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e45a6) [0x7f6b7f0a65a6]
Err: #32  0x7f6b7fb2bb62 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
Err: #33  0x7f6b7fb2e3a1 in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe1
Err: #34  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2521b2) [0x7f6b7fc211b2]
Err: #35  0x7f6b7fc212d1 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf1
Err: #36  0x7f6b7fd109ca in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa
Err: #37  0x7f6b7fb71fae in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ce
Err: #38  0x7f6b7fb3215f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7f
[... cut ...]
Err: #56  0x7f6b7f076c00 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x90
Err: #57  0x7f6b813cb148 in Gui::Application::runApplication() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x2498
Err: #58  /home/user/Programs/FreeCADmaster/build/bin/FreeCAD(+0x98ac) [0x558ae8a4e8ac]
Err: #59  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7f6b7ea2ad0a]
Err: #60  /home/user/Programs/FreeCADmaster/build/bin/FreeCAD(+0x9b5a) [0x558ae8a4eb5a]
```
i second @xtemp09, that this is a duplicate from #8179

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

closed 2024-10-07 by maxwxyz