FreeCAD/FreeCAD #9011: [Issue] Profile Operation does not properly respect Clearace/Safe Heights

ShadeTechnik, 2023-03-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * freecad.gears 1.0.0
  * kicadStepUpMod 10.14.5
  * dxf-library
  * fasteners 0.4.54
  * Assembly4 0.12.6
[/code]
```

### Subproject(s) affected?

Path

### Issue description

Setting Clearance heights and Safe Heights do exactly the same thing, all Z movement is modified by both values. This is different from the Drilling Op that respects Clearance and Safe differently

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-01:
@mlampert would you have a moment to kindly look in to this? TIA!

luzpaz, 2023-06-10:
@ShadeTechnik can you test on dev version ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

blazini36, 2023-11-19:
Still works exactly the same as of this version:
Version: 0.22.0dev.35039 (Git)

Drilling ops and milling ops treat clearance and safety opposite. It's so confusing I don't remember which is which