FreeCAD/FreeCAD #9024: [Issue] Sketcher background color not set via edit/preferences/sketcher/colors.

jopetma, 2023-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
```

### Subproject(s) affected?

None

### Issue description

Sketcher is not usable as background color of part is to light, the sketched items in white color disappears on light grey background as the light grey background could not be changed and color on sketched items stays the same nevertheless I change to in my user prefs.

Either remove the option for user to set background color in edit/preferences/sketches/color or fix the bug that the users preferred background color is not set. I prefer the latter choice. 

This far Im impressed by freecad but it fails on some quite basic functions that should be sorted in early beta release so not yet a valid alternative.

### Anything else?

Anything else...well since youre asking : The pupil in your eye is round, not square 16:9. As your pupil in your eye is round it sees items up and down just as well as side to side. If you understand this simple fact you should realize 16:9 screen format is fraud. As we now are using 16:9 screens I want to place toolbars far out on right side of screen as that space is of less value/usability than top/bottom screen estate. The top and bottom screen estate is extremely valuable as computer screens for stupid sales reasons are 16:9 and top/bottom is smallest space I can use for part design, hence here is where there never should be any toolbars placed ever..soo...I want better possibility to place toolbars out on the useless screen estate.

I guess you wont be asking me again =0)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-03-25:
@jopetma, human eyes are in the position when the principal retinal plane is coplanar with the transverse plane. It is the reason why the picture aspect ratio for displays evolved from 4:3 to 16:9 ― monitors evolve like any other creature.

Most programs, like Word, Excel, MindManager etc utilize UI, in which buttons are placed near the top of a window on purpose.

Your proposal to place the toolbars parasagittally will constrain user to turn his head all the time. Anyway, you always can put toolbars in the way you want, as shown in the video: https://youtu.be/GcgoXgCp-u4?t=653

And you always can customize color scheme as shown in the video: https://www.youtube.com/watch?v=DnWe50PH_GQ

0penBrain, 2023-03-26:
> Sketcher is not usable as background color of part is to light, the sketched items in white color disappears on light grey background as the light grey background could not be changed

Not clear. If the background is another part, you can hide the part or change its color. There is visibility automation to hide parts in some conditions.

> color on sketched items stays the same nevertheless I change to in my user prefs.

Please give precisely which items/colors are concerned.

> Either remove the option for user to set background color in edit/preferences/sketches/color

This option doesn't exist.

> or fix the bug that the users preferred background color is not set. I prefer the latter choice.

??? That's still another point that would deserve details to be understood.

adrianinsaval, 2023-03-28:
The monitor aspect ratio discussion is not relevant here, do it elsewhere. The "Anything else" field is in case you want to add more stuff about the issue, not about some other random unrelated subject.

There is no background color preference in sketcher colors page. The background colors are set globally at _Preferences > Display > Colors_

0penBrain, 2023-04-02:
@jopetma Any follow-up please ?

sliptonic, 2023-04-13:
Can we get an update on this issue.  I don't see the relevant setting to set sketcher background colors

0penBrain, 2023-04-13:
> Can we get an update on this issue. I don't see the relevant setting to set sketcher background colors

I propose to close without further action.

adrianinsaval, 2023-04-15:
I agree with closing since OP has not responded to clarify.

closed 2023-04-15 by adrianinsaval