FreeCAD/FreeCAD #9028: Problem:  Customize Commands doesn't do anything

sliptonic, 2023-03-24:
When the user launches the Customize dialog (tools->customize) The first tab shown is 'commands'.  Nothing on this screen can be changed or customized.   This is confusing.

There may be value in being able to explore the available commands but it really isn't a customization.

Is there any value in preserving this feature?  If so, where should it live to be more intuitive?

[image]

adrianinsaval, 2023-03-28:
yeah this tab should be removed

chennes, 2023-03-28:
Agreed (and @Pesc0 the ampersands show the accelerator key for the command, it's Qt's syntax that for some reason we are displaying to end users 🤷 ).

Pesc0, 2023-03-28:
> the ampersands show the accelerator key for the command

Right, I think that should be removed, it's cumbersome to read and un-intuitive. Perhaps the `keyboard` tab, where accelerator keys are set, can receive a refresh similar to the workbench tab. Also, assuming I didn't misunderstand, it doesn't always work:

[image] 
[image]

On that note: to be honest I would love to see the whole customize menu moved into preferences. I think having all the settings in a single place is good for usability and for new users to discover what's available. I know it took a while for me to discover I could create custom toolbars and move stuff around in the interface, and that's because the toolbar settings were not under the traditional 'preferences' but (hidden) under 'customize' instead. 

And, speaking of settings, further still from the topic of this issue, it's super weird that some settings are visible only when a workbench is loaded. Creates quite a bit of confusion for someone new.

chennes, 2023-03-28:
> And, speaking of settings, further still from the topic of this issue, it's super weird that some settings are visible only when a workbench is loaded.

That was actually the original reason the Workbenches preference page was created -- the idea was that if you went looking for your preferences and didn't see your workbench, your next best option would be to click on "Workbenches" which had a short explanatory text at the top telling you to load your workbench. It helped reduce that particular help question a ton.

Pesc0, 2023-03-28:
Oh, I thought lazy loading was just to speed up application launch. IMO having to resort to 'fallbacks'...
> your next best option would be to click on "Workbenches"

means something is not particularly intuitive about the process, and the user is forced to proceed by trial and error. Something like this would be much more user friendly I think

[image]

adrianinsaval, 2023-03-28:
That's a good idea but then we need a system for addons to register that they have a preference page without loading them. Could be something to add to package.xml maybe? I think you should open a separate issue for this conversation though

Pesc0, 2023-03-28:
Right, moving this to another issue tomorrow. Regarding this topic I see we agree on removing the "Commands" tab.

sasobadovinac, 2023-03-29:
Such Commands tabs in Customize dialogs are normally used to build custom toolbars by drag and drop of the commands, this is by itself actually very common in UI/UX design, but indeed in FC the drag and drop functionality is not working and the editing of toolbars is done from the Toolbars tab.

sliptonic, 2023-04-07:
Any reason not to move ahead with removing the Commands tab?

chennes, 2023-04-07:
It literally does nothing -- I'm good with removing it.

sasobadovinac, 2023-04-07:
My above comment about it should be seen only as informational on how such a commands tab is normally used, since in FC it is actually not working I am not objecting on its removal...

Pesc0, 2023-04-07:
#9110 can be merged then?

PR #9110 (2023-04-07): Remove "commands" tab in customize dialog
Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the features added or changed once this  [... cut]
closed 2023-04-07 by chennes