FreeCAD/FreeCAD #9070: [Problem] "Mirror" means different things in different workbenches.

natewaddoups, 2023-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

In the Part Design workbench, "Mirror" means "create a living copy of this geometry, and keep it updated in the future." The newly created mirrored copy **does** change when the original is changed. This is great!

In the Sketch workbench, mirroring works quite differently.

Expected behavior: same as the Part Design mirror.

Actual behavior: "Mirror" means "create a copy of this geometry, as it exists at this moment." The newly created mirrored copy **does not** change when the original is changed.

### Anything else?

As new user, the difference was surprising. After being disappointed by the behavior of the Sketch mirror, it never would have occurred to me that the Part Design mirror worked the way I was hoping that the Sketch mirror would. 

And, since I'm only creating 2D files for laser cutting, I never would have discovered the Part Design mirror behavior on my own. Fortunately, some kind folks on the forum helped me out with that.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-27:
My personal interpretation is the PartDesign Mirror, Part Mirror and Draft Clone with a negative scale are _Modifiers_ whereas Sketcher Mirror is a _one-time Operation_ and I always expected that to be the case, maybe it's just what I was brought up on. Whether this can be classed as a **Problem** is very dubious, surely it's just an opinion.

On another note, it wouldn't bother me if the Sketcher Mirror was made difficult to access as it promotes poor modelling practice but that's a whole different story.

adrianinsaval, 2023-03-28:
This issue should be refocused to sketcher mirror and change the title to reflect that, other WB don't need any action.

In sketcher the mirror tool should create symmetry constraints between the mirrored geometries. I don't think we need more than that.

natewaddoups, 2023-04-02:
Is there some advantage to that constraint-only approach that you're suggesting?
I mean, why shouldn't the Sketcher mirror work the same way as the Part Design mirror?

adrianinsaval, 2023-04-02:
Ease of implementation and flexibility of the tool.
Making it a mirror like the PD one means creating a whole new geometry type, using constraints is just a matter of tweaking the current implementation.
As for flexibility, using constraints means you can remove some undesired constraints and make a variation of the mirrored shape if desired.

In general, if you want the PD mirror use the PD mirror, patterns in sketches are not the best workflow

natewaddoups, 2023-04-03:
Sketches seem like the ideal workflow for laser cutting projects. There's nothing 3D about the fabrication process, so using 3D features of FreeCad in order to get proper mirror functionality is a weird workflow.

Would it be practical to create a new geometry type that only contains a reference to an existing sketch and logic to invert the X and/or Y coordinates? 

I might be willing to implement this, if someone can point me in the right direction to get started. Are there existing docs I should look at to find the answers to these questions?

Where is the class that implements the existing Sketch type?
Where should I put code for a new MirroredSketch type?
Where is the code that runs when the user clicks OK on the Sketchers "Select Mirror Axis" dialog box? 

I'm hoping I can basically replace the existing code with something along the lines of "return new MirroredSketch(sourceSketch, axis)" ...and then implement the MirroredSketch class as thin wrappers that just invert X and/or Y coordinates on data from the source sketch.  

But of course there are lot of assumptions baked in that idea, beginning with the assumption that FreeCad is written in an OOP language and style.

In cases where a Sketch member function has to return a new object, how are those objects deleted? Is that something I'll need to work on, or can I count on existing object-lifetime-management code in FreeCad to solve that problem?

FlachyJoe, 2023-04-03:
> using 3D features of FreeCad in order to get proper mirror functionality is a weird workflow

Draft is the workbench you look for as it acts in 2D and can deal with sketches. See [Draft_SelectPlane](https://wiki.freecad.org/Draft_SelectPlane) and [Draft_Mirror](https://wiki.freecad.org/Draft_Mirror)

maxwxyz, 2024-04-13:
Is this still relevant? I think this should be solved with the new symmetry tool in sketcher https://github.com/FreeCAD/FreeCAD/pull/12450, please try it in the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

natewaddoups, 2024-04-18:
Thanks @maxwxyz, I'll give that a try.

closed 2024-05-18 by maxwxyz