FreeCAD/FreeCAD #9071: [Issue] In the Docker container, I cannot use freecad to save stp files as images.

Della-Z, 2023-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
version：0.18.4
operating sys：ubuntu 20.04.6 LTS 
64-bit
```

### Subproject(s) affected?

None

### Issue description

I want to use freecad in the docker container to save the stp file as an image using the saveimage method. but When I run the command "Gui.activeDocument().activeView().saveImage" ('/home/rootz/Documents/1.png', 1137, 427, 'White') , it displayed "Directory where to save image doesn't exist" . But I'm pretty sure the directory specified for saving exists.

I see the saveimage source code，The error occurs in the following location of the saveimage method in the src/Gui/View3DPy.cpp file:
if (!fi.absoluteDir().exists())
throw Py::RuntimeError ("Directory where to save image doesn't exist");

So I check the directory permissions with Qfileinfo .iswriteable(), which returns true. , I check the system file permission. The current user has the rwx permission.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-27:
Are you running FreeCAD `version：0.18.4` ?

Della-Z, 2023-03-28:
I've solved the problem today，thanks.
The execution permission of Docker is different from that of VMs.