FreeCAD/FreeCAD #9072: when i click the "print preview" menu and than changed the paper to horizontal the program close itself[Problem]

bilcam36, 2023-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
when i click the "print preview" menu and than changed the paper to horizontal the program close itself
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

file menu

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-28:
please post full version information. At the bare minimum you must say what OS you are using

AjinkyaDahale, 2023-03-29:
Cannot reproduce with 

```
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32499 (Git)
Build type: Release
Branch: core-detect-new-save
Hash: 1e5fa5de6057014518cb89585f7ac575321181f9
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/India (en_IN)
```

xtemp09, 2023-03-29:
I cannot reproduce the crash with the latest stable and the latest master version on Windows 10 and on Ubuntu. Perhaps, it depends on a printer or a file. Though, I can't figure out why the page turned vertical, if it were horizontal and nothing toggled the radiobutton to turn it into portrait layout.

Perhaps, logging will shed some light on this.

luzpaz, 2023-03-29:
@bilcam36 please provide Full [About](https://wiki.freecad.org/About) info please!

luzpaz, 2023-04-10:
@bilcam36 ping

adrianinsaval, 2023-04-14:
I can't reproduce the crash, but the behavior is buggy, the portrait/landscape state of the preview page is not kept in sync. This can also be observed in OP's video

```
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
Installed mods: 
[/code]
```

adrianinsaval, 2023-05-04:
nobody else seems to be able to reproduce and we got no response with further info from @bilcam36, I will close this as it is not actionable as is. Feel free to reopen if you are still experiencing this and can provide us more information.

closed 2023-05-04 by adrianinsaval