FreeCAD/FreeCAD #9086: [Issue] 0.21 Windows 10 : Disabling the splash screen blocks the startup of Freecad

Grubuntu, 2023-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77126

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32532 (Git)
Build type: Release
Branch: master
Hash: 20e44eba50e1365b9e7bb9c5c244afe96a627a28
Python 3.8.16, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

None

### Issue description

If disable splash screen at startup Freecad doesn't start ,it seems that it only happens with  Windows 10 :

1 - Disable splash screen at startup  ( Edit > Preferences > General > Start up settings)

[image]

2 - Close Freecad
3 - Run Freecad again : it doesn't start

-----

4- Re-enable the option by editing the configuration file user.cfg  -> FCBool Name="ShowSplasher" Value="1" 
5 - Run Freecad : it starts

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-03-28:
Last working build I have:

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.21.31155 (Git) Approx 8th Dec 2022
Build type: Release
Branch: master
Hash: 6f1b29769b33676edca2bd883b80bed25a0758c8
Python 3.8.15, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

First non-working build I have:

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31850 (Git)  Approx 13th Jan 2023
Build type: Release
Branch: master
Hash: d8d3bb8120ad139cf1eb2acf0be0ae6c297ac7f3
Python 3.8.16, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

So only 695 commits to plough through but I can't compile Windows.

Syres916, 2023-03-28:
@kzootech Could you please confirm that this change https://github.com/FreeCAD/FreeCAD/commit/2f575a3c91a0b4cd50ac5e7951d2e5f0445f07e3 does not cause the regression mentioned here, TIA.

Syres916, 2023-03-29:
@Grubuntu I've started a git bisect so I should have the definitive root cause by tomorrow (Thursday), it's about an hour per change on my system.

0penBrain, 2023-03-29:
@PaddleStroke @donovaly May one of you bisect this ?

adrianinsaval, 2023-04-15:
> @kzootech Could you please confirm that this change [2f575a3](https://github.com/FreeCAD/FreeCAD/commit/2f575a3c91a0b4cd50ac5e7951d2e5f0445f07e3) does not cause the regression mentioned here, TIA.

I have no clue how you figured that one out but yes, I compiled that one and it's parent 7637d58df0ca64c8975cea09b275feadd7241534

2f575a3c91a0b4cd50ac5e7951d2e5f0445f07e3 does not work with splashcreen disabled, 7637d58df0ca64c8975cea09b275feadd7241534 works

0penBrain, 2023-04-26:
@FreeCAD/maintainers What's the decision about this ?
The regression introduced by the above commit is more severe than the issue it fixes. Shall we then revert ?

commit bed50951 (2023-05-02): Gui: fixes #9086: Disabling the splash screen blocks the startup of Freecad

If no splash screen was shown then windowHandle() returns null and this leads to a segmentation fault.