FreeCAD/FreeCAD #9088: ptogram crashing in techdraw

stew44, 2023-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56
  * Assembly4 0.11.12
  * fasteners
[/code]
```

### Subproject(s) affected?

None

### Problem description

when you try to do various functions the programs crashes completely
it will not let me drop a freecad file in here

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-27:
Zip the FreeCAD file and drag-drop in to GitHub text window

stew44, 2023-03-28:
i tried but it refused all my zip file formats

luzpaz, 2023-04-09:
No response from OP. Closing.

closed 2023-04-09 by luzpaz