FreeCAD/FreeCAD #9095: [Feature Request] Discussion - utilize FEM meshing engines in CfdOF.  And vice versa ?

linuxguy123, 2023-03-28:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?style=5&t=77175

### Subproject(s) affected?

FEM

### Idea description

CfdOF is in need of boundary layer mesh refinement provided by gmsh.   I am throwing around ideas to utilize the FEM meshers in CfdOF.   Here is a diatribe on my thoughts.   https://github.com/jaheyns/CfdOF/issues/78

I have made posts on FreeCAD.org here https://forum.freecad.org/viewtopic.php?style=5&t=77175  and here https://forum.freecad.org/viewtopic.php?style=5&t=77179 and am awaiting input. 

It seems a waste that both workbenches have pursued individual meshers.   Are there hidden issues against merging the meshing for both workbenches ?  

Be aware that I have added remote processing (ie meshing and running the OpenFOAM solver on a remote host) in this repo https://github.com/linuxguy123/CfdOF-RP.  I'm not intending to fork CfdOF, I just forked it to work on it rather than create a branch.  I'll merge the two when the time is right. 

I look forward to your comments. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-03-28:
I don't see this as a FreeCAD issue (yet), what action do you need from FEM developers here? I think it would be a good idea to try to reuse FEM stuff in cfdOf for gmsh but that's something that has to be changed in the addon, not here.

While gmsh is an overlap between FEM and cfdOF, I don't think there's much more beyond that, snappyhexmesh and cfmesh are good for openfoam, but would they be useful for the solvers used in FEM?

linuxguy123, 2023-03-28:
> I don't see this as a FreeCAD issue (yet)

There is no issue, pe se.  There is an opportunity to have a global meshing discussion rather than each party going in their own direction and duplicating efforts. 

> what action do you need from FEM developers here? 

I'm sharing a thought experiment on meshing.   FEM doesn't have cfmesh or snappyHexMesh functionality at this point.  Is there an opportunity to make meshing a global thing ?

CfdOF has some unique requirements in the use of a mesh in that it needs to assign attributes to various surfaces.  This can be done with a generic mesh but it is somewhat more straight forward if that is done as part of the mesh generation process. 

>I think it would be a good idea to try to reuse FEM stuff in cfdOf for gmsh but that's something that has to be changed in the addon, not here.

OK.

> While gmsh is an overlap between FEM and cfdOF, I don't think there's much more beyond that, snappyhexmesh and cfmesh are good for openfoam, but would they be useful for the solvers used in FEM?

In CFD, mesh quality is everything.   More mesh engines equal better meshing options.   For a lot of models with complex geometry snappyHexMesh creates a really good mesh.   Ditto cfmesh.   gmsh, it turns out, has an option (boundary meshing) that is very helpful for thin flow surfaces such as airfoils and heat exchanger fins.  

Upon further consideration, I will be adding boundary layer meshing in gmsh in CfdOF rather than importing a FEM meshes.  The issue with importing a FEM mesh is the matching of surfaces in the FreeCAD model to the patches in the FEM mesh.  It can be done, of course, but the headache probably outweighs the usefulness.

closed 2023-03-29 by linuxguy123

luzpaz, 2023-03-29:
I would have hoped that more FEM folk would weigh-in on this before closing. Hopefully the forum thread will get more love. Thanks for the initiative @linuxguy123 !