FreeCAD/FreeCAD #9105: Orbit around object center Doesn't work, defaults to cursor position[Issue]

fantoche7, 2023-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2004
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * BIM 2021.12.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

Setting the rotation mode of the orbit tool to "object center" doesn't seem to work, it seems to default to orbit around the position of the cursor once i first click the middle mouse button
so if the mouse is out of the objects shape, the moment  I start orbiting that shape jumps out of view

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-03-28:
I don't think I understand this text fragment. Are you refering to https://wiki.freecad.org/Mouse_navigation?

fantoche7, 2023-03-28:
> I don't think I understand this text fragment. Are you refering to https://wiki.freecad.org/Mouse_navigation?

No I am refering to the option in Preferences > Display > Navigation >Rotation Mode : [Object center]

xtemp09, 2023-03-28:
I've just checked, it works on Windows with the latest stable version, just like yours.

fantoche7, 2023-03-28:
Here is a screeshot.  am I doing something wrong, or is it a bug

https://user-images.githubusercontent.com/129213099/228338498-c9191ab2-598b-4dba-92d8-71b01b9f4597.mp4

xtemp09, 2023-03-29:
When you set rotation around the object, rotation goes around the object. Click with the middle mouse button sets the center of the viewport. But **rotation goes around the object**.

fantoche7, 2023-03-29:
This is what I believe orbit around object center should behave, the screenshot is from revit
you don't need to set the orbit center using the middle mouse button, once a object is selected the orbit should keep the object in its position (no matter where the middle mouse button happens ) in the viewport and orbit the whole model around it.
[image]

xtemp09, 2023-03-29:
Ok, you're right. I just didn't understand without the image.

luzpaz, 2023-04-01:
> I've just checked, it works on Windows with the latest stable version, just like yours.

@xtemp09 please post full about info please? :pray:

xtemp09, 2023-04-02:
@luzpaz, it's not me who has to report the version info, but the topicstarter. I just didn't understand him. fantoche7's already reported, so nothing to do.

luzpaz, 2023-04-02:
Sorry! I meant @fantoche7

fantoche7, 2023-04-02:
@luzpaz  Is there anything missing about my report of this bug ?

luzpaz, 2023-04-02:
All good. Never mind.

luzpaz, 2024-02-11:
@fantoche7 Do you mind checking if this is still relevant for latest development version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

luzpaz, 2024-03-11:
@fantoche7 :point_up_2:

zihotki, 2024-05-25:
@luzpaz  unfortunately it's still broken. 

OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Part-o-magic 1.1.0
  * freecad.gears 1.2.0
  * woodworking 0.21.2.33771
  * A2plus 0.4.64a
  * BIM 2021.12.0

My settings:
[image]

When I try to rotate an object, it rotates at a pivot point at the center of the screen
[image]

Rexbas, 2024-05-25:
Looks like this only happens to Gesture and MayaGesture navigation styles. When using CAD navigation style it works as expected.

capkoh, 2024-06-03:
Hi, I also faced this issue. **Rotation Mode > Object Center** does not work to me as expected.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
```
Checked different navigation styles, all of the following behave incorrectly: CAD, Blender, TinkerCAD. I suppose all navigation styles are affected by this issue.

closed 2024-06-10 by chennes