FreeCAD/FreeCAD #9112: Freecad startet nicht in Windows 10

sigigros, 2023-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Dateiversion 020.2.2
```

### Subproject(s) affected?

None

### Issue description

Hallo, ich habe FreeCad 0.20.2.2 auf meinem Laptop installiert, jedoch startet das Programm nicht.
Nach Doppelklick auf das FreeCad Symbol sieht man einen Startvorgang für etwa 5 Sekunden, dann geschieht nichts weiter.
Ich würde gerne auch etwas spenden, aber für ein Programm, welches nicht einmal startet ...
Ich hatte vorher Autocad Inventor 2020 auf meinem Laptop installiert - dieser lief einwandfrei ...

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-03-29:
https://wiki.freecad.org/Debugging recommends running FreeCAD with an option `-l [ --write-log ]` to write logs.

adrianinsaval, 2023-03-29:
Please write in english

sigigros, 2023-03-29:
OK: I have installed FreeCad 0.20.2.2 on my Laptop. But the Software doesn`t start.
If I doubleclick on the FreeCad Icon, it starts to load but only about 5 seconds. Then nothing else happens.
I would even donate for FreeCad, but if the Software doesn´t even start - sorry.
Befor FreeCad I had installed >Autocad Inventor 2020< , it was running perfect. But the Licence has expired.
So can anybody tell me what´s the problem - even better - who can solve it?
Thx

luzpaz, 2023-03-29:
Try installing the development version to see if you can reproduce the issue
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

adrianinsaval, 2023-03-29:
check the known issues list and see if one of those applies to you: https://forum.freecad.org/viewtopic.php?t=30573

it's likely a graphics driver issue

adrianinsaval, 2023-03-29:
also do as xtemp09 has said, run `FreeCAD.exe --write-log` and provide the log file that should be created at %APPDATA%/FreeCAD
without more information from you there's nothing we can do.

luzpaz, 2023-04-01:
@sigigros ping

luzpaz, 2023-04-02:
Closing since OP is unresponsive.

closed 2023-04-02 by luzpaz

sigigros, 2023-04-04:
I have solved the issue.
If there are two graphiccards on the Notebook and one of it is an AMD mobility -
open the "catalyst center" 
go to "umschaltbare Grafikkarten"
go to "Anwendung hinzufügen"
and choose "Path\FreeCad.exe" on your System.
click on "FreeCad.exe" and choose "Hochleistung" (in the attached Picture it´s another Program - example like "videodeluxe.exe"
click on "save" (übernehmen)
and so FreeCad should start without any issue.
[image]

luzpaz, 2023-04-05:
Thanks for the follow-up!