FreeCAD/FreeCAD #9114: [Issue] Unable to engrave an arc.

adrianre12, 2023-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77106

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32457 (Git)
Build type: Release
Branch: master
Hash: 85216bd12730bbc4c3cbf8f0bc50416ab1556cbb
Python 3.8.16, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

Path

### Issue description

I first found the issue in 0.20 and after a discussion on the forum, I can also reproduce it on 0.21.
The overall issue is that I was trying to engrave a faceplate dial that consisted of an arc with graduations and numbering. What I found is that it would calculate paths for everything except the arc.

I have since created a simpler example that shows the issue and what I have found.

- I select the job and press engrave so it auto-selects all the models, the path for the arc will not be created. (Engrave1)

- Press engrave and manually add each model to the base geometry, the path for the arc will not be created. (Engrave2)

- Press engrave and manually add the models for A, line, square, and circle, then select the line of the arc and add it. Only the path for the arc will be generated (Engrave3) If the Model-Arc.Edge1 is removed, the paths for the A, line, square and circle will be generated.

- Press engrave and manually add the model for the A, and then select and add each line segment for the line, square, circle and arc. The paths for line, square, circle and arc will be generated but the ShapeString A will be omitted. (Engrave4)If lines in the A are selected instead of the model they will be added to the path.

It seems that the models in the base geometry are ignored if any line segments are present.
The arc model does not automatically get processed.
[Example2.zip](https://github.com/FreeCAD/FreeCAD/files/11099266/Example2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Russ4262, 2023-06-10:
This issue involves to bugs.  One affects combination of edges and entire models as Base Geometry selections.  The other affects certain edges.

luzpaz, 2023-06-10:
@Russ4262 do we have open tickets for them already ?

Russ4262, 2023-06-10:
I don't speak for project managers or maintainers, but my opinion is that this ticket is serving as such.

Also, thanks for the simplified file you provided. It was very helpful in identifying the bugs.

adrianre12, 2024-10-15:
Did this get fixed?

luzpaz, 2024-10-15:
By the looks of it, no.

JohnOCFII, 2025-01-14:
@maxwxyz  - this is still a live issue that many are waiting to see resolved.

tarman3, 2025-07-04:
We should send wires to `buildpathocc()`
but Shape of `Arc` object is `Edge` and list of wires is empty

```
>>> obj.Shape
<Edge object at 0x5589b6fb5630>
>>> obj.Shape.Wires
[]
```

https://github.com/FreeCAD/FreeCAD/blob/927d302401caad6a9e518584e0665ae9f61509f2/src/Mod/CAM/Path/Op/Engrave.py#L138-L144

Probable fixed this issue in #22226
https://github.com/FreeCAD/FreeCAD/blob/14a914a3b2ff15c6bad92be5a0ad875fe1d4b871/src/Mod/CAM/Path/Op/Engrave.py#L165-L168