FreeCAD/FreeCAD #9115: [Issue] TopoNaming Phase 2 (card 1 of n) - create ElementMap.cpp (and more to ElementMap.h)

JohnAD, 2023-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.14555 (Git shallow) Snap 222
Build type: Debug
Branch: master
Hash: d4feb6578398d51eaf56cd794e8d3e2e000bea64
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

> The toponaming algorithm needs a place to express the generatively express the history of an item.

## description

This class embodies the postfix support in `src/App/ComplexGeoData.cpp`

It is a serializable sorted mapping of items which generatively describe the elements and it's history.

The unit testing should mock up a variety of scenarios that would be seen in real-world projects and assert expected strings. This is critical as future changes to the class should be as consistent and non-breaking as possible.

This step also adds the `elementMap`/`_elementMap` members seen in the the ComplexGeoData class; though it is left empty when FreeCAD is running.

For context, see issue #8432

## notes

Please don't confuse the ElementMap class with the MappedElement class, which stores the IndexedName and MappedName data items.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-30:
I'll grab this one and start in on it.

chennes, 2023-03-30:
(This is on hold for me while I work on card 1, so of someone else wants it feel free to pick it up.)

Pesc0, 2023-04-03:
@JohnAD Looks like ElementMap depends on some functions and members of ComplexGeoData. 

- `findTagInElementName`
- `App::StringHasherRef Hasher`
- `hashElementName`
- `encodeElementName`
- `setElementName`

Some of these functions should be defined in card 3 #9116
How should I proceed?
The following is the build log of the PR I opened 

```
 /home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:503:52: error: no member named 'findTagInElementName' in 'Data::ComplexGeoData'
        if (name.find("#") >= 0 && ComplexGeoData::findTagInElementName(name) < 0) {
                                   ~~~~~~~~~~~~~~~~^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:512:13: error: 'this' argument to member function 'compact' has type 'const Data::MappedName', but function is not marked const
            ret.first->first.compact();
            ^~~~~~~~~~~~~~~~
/home/runner/work/FreeCAD/FreeCAD/src/App/MappedName.h:720:10: note: 'compact' declared here
    void compact();
         ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:567:29: error: no member named 'findTagInElementName' in 'Data::ComplexGeoData'
        if (ComplexGeoData::findTagInElementName(
            ~~~~~~~~~~~~~~~~^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:715:42: error: no member named 'Hasher' in 'Data::ComplexGeoData'
    if (childElements.empty() || !master.Hasher)
                                  ~~~~~~ ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:723:39: error: no member named 'findTagInElementName' in 'Data::ComplexGeoData'
            int pos = ComplexGeoData::findTagInElementName(
                      ~~~~~~~~~~~~~~~~^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:726:45: error: no member named 'hashElementName' in 'Data::ComplexGeoData'
                MappedName postfix = master.hashElementName(
                                     ~~~~~~ ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:731:24: error: no member named 'encodeElementName' in 'Data::ComplexGeoData'
                master.encodeElementName(
                ~~~~~~ ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:842:20: error: no member named 'encodeElementName' in 'Data::ComplexGeoData'
            master.encodeElementName(
            ~~~~~~ ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:872:24: error: no member named 'encodeElementName' in 'Data::ComplexGeoData'
                master.encodeElementName(
                ~~~~~~ ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:874:24: error: no member named 'setElementName' in 'Data::ComplexGeoData'
                master.setElementName(idx, name, &sids);
                ~~~~~~ ^
/home/runner/work/FreeCAD/FreeCAD/src/App/ElementMap.cpp:890:20: error: no member named 'encodeElementName' in 'Data::ComplexGeoData'
            master.encodeElementName(
            ~~~~~~ ^
```

Pesc0, 2023-04-03:
Question 2:
Currently in `ElementMap.h` I have a big block of 
```
static constexpr const char* POSTFIX_TAG = ";:H";
static constexpr const char* POSTFIX_DECIMAL_TAG = ";:T";
static constexpr const char* POSTFIX_EXTERNAL_TAG = ";:X";
static constexpr const char* POSTFIX_CHILD = ";:C";
static constexpr const char* POSTFIX_INDEX = ";:I";
static constexpr const char* POSTFIX_UPPER = ";:U";
static constexpr const char* POSTFIX_LOWER = ";:L";
static constexpr const char* POSTFIX_MOD = ";:M";
static constexpr const char* POSTFIX_GEN = ";:G";
static constexpr const char* POSTFIX_MODGEN = ";:MG";
static constexpr const char* POSTFIX_DUPLICATE = ";D";
```
Compare this to the following functions of ComplexGeoData:
```

const std::string &ComplexGeoData::tagPostfix() {
    static std::string postfix(elementMapPrefix() + ":H");
    return postfix;
}

const std::string &ComplexGeoData::decTagPostfix() {
    static std::string postfix(elementMapPrefix() + ":T");
    return postfix;
}

const std::string &ComplexGeoData::externalTagPostfix() {
    static std::string postfix(elementMapPrefix() + ":X");
    return postfix;
}

const std::string &ComplexGeoData::indexPostfix() {
    static std::string postfix(elementMapPrefix() + ":I");
    return postfix;
}

const std::string &ComplexGeoData::missingPrefix() {
    static std::string prefix("?");
    return prefix;
}
```

Where should these definitions live? 
ComplexGeoData::elementMapPrefix() is referenced in many places, should this live with the rest of the definitions or it has its own place?

JohnAD, 2023-04-04:
Just using `encodeElementName` for a first look:

> `ComplexGeoData` class
> has `setMappedChildElements`
> which calls `ElementMap`->`addChildElements` with (another?) `ComplexGeoData` as a parameter

and

> `ElementMap` class
> has `addChildElements`
> which calls `ComplexGeoData::encodeElementName`

So ComplexGeoData refs ElementMap refs ComplexGeoData ....

While this isn't strictly forbidden, I'm going to think of a safer less-recursive way to lay out those class methods.

JohnAD, 2023-04-04:
As to the many tag name functions; I was thinking it made more sense for them to referenced by the named constants rather than having functions return constants. If you want, the elementMapPrefix could be made a constant also; but that is up to you. Again, it is simply returning a constant string literal. On the other hand, we aren't trying to do too much refactoring.

JohnAD, 2023-04-05:
It is good to have a challenge :).

The first one is simple: please move 

```cpp
int findTagInElementName(const MappedName & name,
                                   long *tag=0,
                                   int *len=0,
                                   std::string *postfix=0,
                                   char *type=0,
                                   bool negative=false,
                                   bool recursive=true);
```

to the `MappedName` class and add a test.

I'm not sure why this was in `ComplexGeoData`. Perhaps for easy access to the tag string functions like `tagPostfix()`?

Perhaps those string constants should be in a reference file by themselves to prevent circular references. `PostfixStringReferences.h` perhaps?

You can make it a basic method. Or, for now, you can make it a static function. I suspect this will also break a python interface, so that needs to be addressed also.

By the time I've figured out `encodeElementName` I suspect `hashElementName` and others will be moved elsewhere as well?

@chennes does this sound like a sane direction?

Pesc0, 2023-04-05:
I actually had already done that! I did not push because I wasn't sure it was the right way. 

Unrelated: can we put all the files we created so far in a new folder `src/App/Toponaming`? Just to keep things organized

Pesc0, 2023-04-05:
I've moved things around a bit, I still need to add the test but I've run out of time for today.

JohnAD, 2023-04-16:
Examining more of the code:

* `hashElementName`

  move the following from `complexgeodata.h` to `elementmap.h`:

    `typedef QVector<App::StringIDRef> ElementIDRefs;`

  add `App::StringHasherRef hasher` (was `this->Hasher`) to the parameters

  and then move the entire procedure to `elementmap.cpp`

* `encodeElementName`

  simply move to `elementmap.cpp`

  This _should_ work, assuming that the `ElementIDRefs` typedef was also moved to `elementmap.h`

* `setElementName`

  move to `elementmap.cpp` and add `ElementMapPtr &elementMap` to the parameters, which replaces to the local class reference of `_ElementMap`

  when it calls `resetElementMap`, instead do:

    `elementMap.swap(std::make_shared<ElementMap>())`

  (I may have messed up my pointers just now. Let the compiler guide you. :) )

  but also move:

    `renameDuplicateElement` to `elementmap.cpp`

JohnAD, 2023-04-16:
Also, I've free for a 24-hour sweep of time this coming Tuesday. We could pair-program on this if you would like via Slack/Zoom/VSCode/etc. (Anything other than WebEx; WebEx does not like my linux machine.) Just ping me on Telegram with a good time that works for you.

Pesc0, 2023-04-17:
Done, now it builds, thank you for helping! 
TODO list
- reorder things a bit
- write tests
- refactor/cleanup

Regarding pair-programming I may not be the best candidate: I'm new here and have very little experience (with the codebase and in general). But if someone else is willing to (maybe @chennes) I would love to join and learn something new!

JohnAD, 2023-04-30:
I've been looking through the code this weekend. To help, I've done a PR to the branch your PR is based on. This change simply removes the recursive reference back to the ComplexGeoData class. I'm not sure of the `masterTag` name; perhaps `parentTag` is a better name. Passing a simple long rather than a pointer should have almost the same performance impact.

PR #9175 (2023-06-15): [Toponaming] create ElementMap class
Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the features added or changed once this PR is [... cut]
closed 2023-06-15 by chennes