FreeCAD/FreeCAD #9127: [Issue] TopoNaming Phase 2 (card 1 of 5) - transfer StringHasher

JohnAD, 2023-03-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.14555 (Git shallow) Snap 222
Build type: Debug
Branch: master
Hash: d4feb6578398d51eaf56cd794e8d3e2e000bea64
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

> The toponaming algorithm needs to hash strings in a specific manner.

## description

Transfer the StringHasher.h/.cpp files to master. Refactor as needed.

For context, see issue #8432

## notes

Used by the ElementMap struct functions.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-30:
I've started in on this one, minus the DocumentParams stuff: not that there is anything wrong with that code, but it didn't really seem related to Toponaming.

chennes, 2023-03-31:
`StringHasher` is also going to require the binary read/write code be added to Writer.* and Reader.* -- I am _not_ planning on modifying `InputStream` or `OutputStream` to support non-binary data, however, it looks to me like we can skip that bit of code addition.

chennes, 2023-04-03:
The `StringID` class provides a `toString` function and a `toBytes` function that behave quite differently from one another. That doesn't seem to be necessary, however: `toBytes` only appears to be called once, and could easily have its implementation match that of `toString`. @realthunder or @JohnAD is there any reason not to do this?

JohnAD, 2023-04-04:
@chennes That sounds reasonable to me. We can always add an additional function later if really needed.

realthunder, 2023-04-04:
@chennes Looks fine to me too.

chennes, 2023-04-05:
Another one for you guys: `operator<` in `StringIDRef` currently begins like this:
```
bool operator<(const StringIDRef& stringID) const
    {
        if (!_sid) {
            return true;
        }
        if (!stringID._sid) {
            return false;
        }
```
This creates an inconsistency, in that the sense of the operation flips depending on if you are doing a<b or b<a when both a and b are unattached from a `StringID`. It seems like if both `_sid` and `stringID._sid` are `nullptr`, then the comparison should be `false` (because they are effectively equal), right?

realthunder, 2023-04-05:
> It seems like if both `_sid` and `stringID._sid` are `nullptr`, then the comparison should be `false` (because they are effectively equal), right?

Yes, you are correct. My mistake.

> `_sid` is a pointer, so it's checking whether the two refs refer to literally the same `StringID` object, not whether the two SIDs are equal to one another

`StringID` instances are supposed to be uniquely generated by `StringHasher`, so it is enough to only compare the pointer value. It does not make much sense mixing `StringID`s from different `StringHasher`s, which belong to other documents/objects, because even if they contain the same textual content, they are referring to different geometry.

realthunder, 2023-04-06:
On second look, the implementation of operator `<` was original similar to `=`, i.e., to compare pointer value. I don't recall exactly the reason of the change, but most likely is to get a more consistent sorting order.

chennes, 2023-04-06:
It's obviously not a big deal, it just caught my eye while I was writing the tests. It seems to me that it would be best if we set it up so that it was not possible to have all of `a < b == false`, b < a == false` and `a != b`, which may lead to nonsensical results in future unanticipated operations.

realthunder, 2023-04-06:
Understand. I have no objection to your change of operator `==`.

chennes, 2023-04-07:
@realthunder where is the 10 coming from here:
```
unsigned int StringHasher::getMemSize() const
{
    return (_hashes->SaveAll ? size() : count()) * 10;
}
```

chennes, 2023-04-07:
@JohnAD, @AjinkyaDahale, @realthunder -- can one of you guys help me understand what this block of code is supposed to do? I'm writing tests blind right now, and I'd rather understand what it is I'm testing!
https://github.com/FreeCAD/FreeCAD/blob/140236c7747aecaedd7e199a05ee81c1081378d9/src/App/StringHasher.cpp#L331-L347

realthunder, 2023-04-10:
@chennes 

The `StringHasher` can be considered as a table for storing unique strings. Although it works with any string, it is mostly optimized to store encoded element names (i.e. `MappedName`). I'll use a simple example to illustrate how element names are encode and stored in `StringHasher`.

Let's start from a box. None of its element name is encoded. So we have Face1, Face2, etc. When it is fused with another shape, one of its changed face will be named as `Face3;:M;FUS;:Hb:7,F`. Here, `Face3` is the source element name of the box, called as the `prefix` of the name and stored in `MappedName::data`. The rest of the name, `;:M;FUS;:Hb:7,F`, is store in `MappedName::postfix`. `StringHasher` is not used at the current stage. Well actually it is used to shorten names for those elements that are generated/modified from more than one elements (i.e. more than one source element), but it is out of the scope of this illustration.

Now, when the fusion shape is again fused with a third shape, the above face is changed again and now named as `#38;:M;FUS;:Hd:7,F`. The prefix `#38` is a `StringID` stored inside a `StringHasher`, obtained by calling `StringHasher::getID(MappedName)`. The 38 here is a hex number, acting as a key mapping to a string that is the source element name, `Face3;:M;FUS;:Hb:7,F`. If `StringHasher` were to naively store the string as it is, then each element name of the source shape will be a unique entry in the table, because the postfix contains a unique ID of each object, for history tracing purpose. Instead, `StringHasher` sotres the prefix and postfix separately similar to `MappedName`.

For prefix, if it is an `IndexedName`, e.g. `Face3` in this case, then only the element type string `Face` will be stored in `StringID`, while the index is stored in the returned `StringIDRef` in `StringIDRef::_index`. Flag `StringID::Indexed` is used to mark this case. The encoded prefix is stored in the related string ID vector, `StringID::_sids`. Note that this optimization is causing some problem during recompute and is later [restricted](https://github.com/FreeCAD/FreeCAD/blob/140236c7747aecaedd7e199a05ee81c1081378d9/src/App/StringHasher.cpp#L266) to only names without postfix. I haven't got time look deep into it yet.

For postfix, if it is not already encoded (i.e. contains no `StringID`), it will be stored in `StringHasher` by calling `StringHasher::getID(QByteArray)` and the returned string ID put in the related string ID vector, `StringID::_sids`. Flag `StringID::PostfixEncoded` is used to mark this case. And `StringID::_postfix` will be assigned the text form of the `StringID`.

Finally, we get to the code you are referring to. For that, let's consider an extra fusion operation, which changes the face `#38;:M;FUS;:Hd:7,F` into `#71;:M;FUS;:Hf:7,F`. `#71` is the string ID that refers to the string `#38;:M;FUS;:Hd:7,F`, where `#38` is the prefix, and the rest is the postfix. For prefix that is also a `StringID`, it is stored in the related string ID, `StringID::_sids`, and marked by flag `StringID::PrefixID`. If the prefix is not only a `StringID`, but also with extra index (e.g. as mentioned above, `Face3` will only store `Face` as string ID, say `#2`, with index stored in `StringIDRef`, represented in text form as `#2:3`), then it will be marked by flag `StringID::PrefixIDIndex`. In both cases, the code will try to put the existing `StringID` in a predefined position in the related string ID vector, for saving/restoring purpose.

chennes, 2023-04-10:
@realthunder thanks for the detailed example: that will definitely help me write the tests 😄 . What about the *10 in `getMemSize()` -- where is that coming from?

realthunder, 2023-04-10:
Well, I consider `getMemSize()` is just a quick hint of the storage size required.

chennes, 2023-04-10:
For my own sanity, the start of a usage diagram (arrows indicate use, not subclassing... I'm new to Mermaid so I don't know how else to build this)
```mermaid
classDiagram
    MappedElement <|-- IndexedName
    MappedElement <|-- MappedName
    class IndexedName{
        const char * type
        int index
    }
    class MappedName{
        QByteArray data
        QByteArray postfix
    }
    class MappedElement{
        IndexedName index
        MappedName name
    }
    class StringID{
        long _id
        QByteArray _data
        QByteArray _postfix
        StringHasher* _hasher
        mutable Flags _flags
        mutable QVector<StringIDRef> _sids
    }
    StringHasher <|-- StringID
    StringHasher <|-- MappedName
    class StringHasher{
        std::unique_ptr<HashMap> _hashes
        StringIDRef getID(const QByteArray& data, Options options)
        StringIDRef getID(const Data::MappedName& name, const QVector<StringIDRef>& sids)
        StringIDRef getID(long id, int index = 0) const
        StringIDRef getID(const StringID::IndexID& id) const
    }
```

realthunder, 2023-04-13:
Yes, you're right. Because `id._postfix.size()>0` was added later for a quick fix. I need to take a deep look at it to see if it is really necessary.

berniev, 2023-04-14:
I don't envy you on this one.

The getID function has a cognitive complexity of 47. Way high. There are so many paths through this code that comprehension and testing are horrible.

Observation 2c FWIW: The function is probably doing way too much but maybe the task is just complicated. However, there is an aweful lot of "Ask" going on. Classes climbing in and out of each other's innards. Refer "TellDon'tAsk "rule. Functionality can be refactored out to where it belongs (i.e. the places being asked). This allows testing small functions in those other classes. These small functions are simpler too as they are dealing directly with local stuff. It may be possible to remove much clutter from the fundamental algorithm.

berniev, 2023-05-07:
Just for fun ... GPT4 suggests (I didn't check or test):
```
StringIDRef StringHasher::getID(const QByteArray& data, Options options) {
    const bool binary = options.testFlag(Option::Binary);
    const bool hashable = options.testFlag(Option::Hashable);
    const bool nocopy = options.testFlag(Option::NoCopy);

    if (hashable && _hashes->Threshold > 0 && data.size() > _hashes->Threshold) {
        QCryptographicHash hasher(QCryptographicHash::Sha1);
        hasher.addData(data);
        const QByteArray hashedData = hasher.result();

        auto it = _hashes->left.find(&hashedData);
        if (it != _hashes->left.end()) {
            return StringIDRef(it->first);
        }

        if (!nocopy) {
            return StringIDRef(insert(StringID(lastID() + 1, hashedData, StringID::Flag::Hashed)));
        }
    }

    const QByteArray copiedData = nocopy ? data : QByteArray(data.constData(), data.size());
    auto it = _hashes->left.find(&copiedData);
    if (it != _hashes->left.end()) {
        return StringIDRef(it->first);
    }

    return StringIDRef(insert(StringID(lastID() + 1, copiedData, binary ? StringID::Flag::Binary : StringID::Flag::None)));
}
```

```
 Improvements in this version include:

    Avoiding unnecessary variable assignments: Instead of assigning the boolean values to separate variables, they are used directly in the conditions.

    Early return: When the data is hashed and found in the hash map, it returns early without proceeding further.

    Avoiding redundant hashing: If the data is hashed and not found in the hash map, it checks the nocopy flag and directly creates a new StringIDRef without rehashing the data.

    Using copiedData instead of modifying dataID._data: In the original code, dataID._data was modified to make a deep copy. However, in this version, we create a new QByteArray called copiedData to store the copied data, leaving dataID._data unchanged.

    Consolidating duplicated code: The code for finding the hashed data in the hash map is consolidated to avoid duplication.

These improvements help optimize the code for better readability, performance, and maintainability. Please note that without the complete context of your code and requirements, there may be further optimizations or adjustments that could be made.
```

PR #9148 (2023-05-16): App/Toponaming: String hasher
This PR will close #9127 -- it is mostly focused on StringHasher, but to pull that code into master it also requires some modifications to Reader, XMLWriter, and Bitmask.

I've marked this as ready for review: there are still a few missing tests related to serialization, but we can consider merging it without those in the interest of getting the other cards moving.

---

- [X]  This Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
closed 2023-05-16 by chennes