FreeCAD/FreeCAD #9130: [Issue] Remove unused files in Mod/PartDesign/FeatureHole

Pesc0, 2023-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
--
```

### Subproject(s) affected?

None

### Issue description

As pointed out in #9103 there might be unused files in Mod/PartDesign/FeatureHole, namely `TaskHole.py` and probably `HoleGui.py`. A thorough analysis is required to find and remove such files. 

It would be a good idea to discuss whether this search should be extended to other folders as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pesc0, 2024-12-27:
@kadet1090 I remember seeing some discussion about a new hole tool? Maybe this has been addressed already?

maxwxyz, 2025-06-11:
@alfrix FYI

Pesc0, 2025-08-17:
still not done afaik

FEA-eng, 2025-10-18:
@Pesc0 I assume this is still relevant

Pesc0, 2025-10-18:
I did a quick check and cannot find the files, perhaps they have been moved but not sure where. Needs to be investigated more to make sure.
Or maybe this has been fixed but the issue has not been closed.

closed 2025-10-19 by maxwxyz