FreeCAD/FreeCAD #9135: [Issue] Sketcher toolbars missing/appear on other workbenches

PaddleStroke, 2023-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=671911#p671911

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32641 (Git)
Build type: Release
Branch: sketcher_toolbar_fix
Hash: 5329cede7f926d55aec396b8fee091889ef14767
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

1. reset config, restart FreeCAD
2. set startup workbench to Sketcher. Restart FreeCAD.
so far so good, Sketcher workbench is loaded, with almost no toolbars
3. make a new sketch
4. while the sketch is still open, switch to another workbench, then back to sketcher
-> no geometry and constraints toolbars, can't do anything.
Reopening the sketch helps.

and then

5. restart FreeCAD
now, all geometry and constraints toolbars appear... which is not normal, but nothing bad really.
6. switch to Part workbench
now, all sketcher geometry and constraints toolbars are on Part! completely displacing all the Part workbench tools.
Switching back to Sketcher and then back to Part fixes things.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-03-31:
The first thing to note is that this issue happen when the user changes of workbench while being in edit mode. In this case, the toolbars are not the only weird thing, the UI looks pretty buggy: 
[image]

First question : Do we want to keep this buggy behavior ? My first thought was that leaving sketcher should close edit-mode. As it doesn't make sense to be in edit mode while in PartDesign or whatever else.

To this one valid counter argument was given : 
- When user change of document tab, either to looks something in another document or in a spreadsheet, then he comes back and want to keep editing his sketch.

So my thoughts are : 
a - If wb changed through the selector should exit the edit mode. 
b - If wb changed because the document tab changed, then the edit mode should not be close.

What do you think about this @abdullahtahiriyo?

PaddleStroke, 2023-03-31:
Secondly, to fix this issue, in sketcher workbench.cpp we need to be able to know if we are in edit mode or not. Or alternatively, in VPSketch we need to be able to know when workbench switched.

So either something like this : 
```
Gui::ToolBarItem* Workbench::setupToolBars() const
{
    Gui::ToolBarItem* root = StdWorkbench::setupToolBars();

    Gui::ToolBarItem* sketcherEditMode = new Gui::ToolBarItem(root, isEditMode() ? Gui::ToolBarItem::HideStyle::VISIBLE : Gui::ToolBarItem::HideStyle::FORCE_HIDE);
    sketcherEditMode->setCommand("Sketcher Edit Mode");
    addSketcherWorkbenchSketchEditModeActions(*sketcherEditMode);
```
but I think this is not possible if I'm not mistaken @wwmayer ?

The other way would be to have a function in vpsketch that is called when workbench is changed. Is that possible?

wwmayer, 2023-04-01:
https://github.com/FreeCAD/FreeCAD/pull/9150 partially fixes the issue. When changing from Sketcher to another wb and back and if the sketch is in edit mode it will restore the toolbars.

The reason for the other problem is that the current state of all toolbars is saved when leaving edit mode. But because the toggle action is visible the function ToolBarManager::saveState() also saves the state of the editing toolbars. Later when quitting the application the toggle actions are invisible again and thus ToolBarManager::saveState() ignores the new state. This means that the user.cfg has stored the state when the editing toolbars were visible which isn't necessarily the last state.

wwmayer, 2023-04-01:
1faf5e5 fixes the second part of the problem.

@PaddleStroke please test if everything works for you now. Because your user.cfg has already saved the visibility state of toolbars it shouldn't have saved it's best to start with an empty user.cfg again.

PaddleStroke, 2023-04-01:
@wwmayer I just build your 9150 and cleared preferences. The second problem is solved but on my side the first problem still occurs. If I change wb when I'm on edit mode, then go back to sketcher, then the toolbars are still wrong : only non-edit toolbar.

wwmayer, 2023-04-02:
> The second problem is solved but on my side the first problem still occurs.

That's weird! For me both issues are properly fixed now. The solution for the first issue is to override Workbench::activated, check for a sketch in edit mode and show the editing toolbars if needed.

Can you set a break point in SketcherGui::Workbench::activated and check what happens there?

PaddleStroke, 2023-04-02:
Yes, I will test in more details tomorrow

PaddleStroke, 2023-04-02:
@wwmayer After some more tests and rebuilding it now works as it should. Not sure what was wrong yesterday but now I can't reproduce the bug anymore. Sorry for the false warning!

wwmayer, 2023-04-03:
Glad that it works now. So, can we merge the PR?

PaddleStroke, 2023-04-03:
Well that's for you to say :)
For me it works great and I can't find any obvious issue in the code (though I'm not a great reviewer yet).
Thanks

closed 2023-04-03 by wwmayer