FreeCAD/FreeCAD #9158: [Issue] sqrt with unit mm doesn't seem to work

ctron, 2023-04-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Expressions

### Issue description

I have a value of  `15mm` with an alias of `foo` in a spreadsheet. I want to use `sqrt(<<parameters>>.foo)`.

The result is an error: `<Spreadsheet> Cell.cpp(645): myfile#Spreadsheet.H14: All dimensions must be even to compute the square root.
in expression: sqrt(15mm)`

This can also be reproduced by just using `sqrt(15mm)`.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-04-01:
What unit would you expect to result from taking the square root of a length?

xtemp09, 2023-04-03:
`sqrt(15 mm^2)` gives correct result.

chennes, 2023-04-03:
I'm going to close this as I wouldn't expect it to work: taking the square root of an area makes sense, and yields a length. But taking the square root of a length yields some non-physical unit, and I believe FreeCAD is right to reject it.

closed 2023-04-03 by chennes

ctron, 2023-04-10:
So assuming I have a spreadsheet with a length (in mm), and want to calculate the diagonal of a square, how would I do this?

Normally I would use: `d = sqrt(2 * <<parameters>>.side)` But if `side` has a unit (like mm), it won't work in FreeCAD.

chennes, 2023-04-10:
That's not the right equation: starting from a^2 + b^2 = c^2, you need 2 * sqrt(2 * <<parameters>>.side * <<parameters>>.side).

mfro0, 2023-04-10:
> Normally I would use: `d = sqrt(2 * <<parameters>>.side)` But if `side` has a unit (like mm), it won't work in FreeCAD.

to get rid of any unwanted unit in an expression just divide by it:
`5 mm / 1 mm = 5`

ctron, 2023-04-12:
Which brings me back to my original report, that it is not necessary in this case.

But as @mfro0 already suggested, there is a way to work around the (IMHO unnecessary limitation) of FreeCAD.