FreeCAD/FreeCAD #9170: [Issue] CI: GTest failures show the wrong completion message and icon

chennes, 2023-04-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Issue description

When a Google Test fails, the CI correctly reports that as a failure in the final analysis, but still shows the message "GTest C++ unit test suite succeeded" with the "success" check mark in the Summary report.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-16:
Hi @berniev do you mind taking a look at this?

berniev, 2023-04-16:
I think @OpenBrain is the expert on CI stuff?

chennes, 2023-04-16:
@0penBrain (with a zero for an O, just to confuse us all 😄 ).

berniev, 2023-04-16:
> just to confuse us 

Certainly worked! I gave up after a while, confident someone would put it right. That worked too!

0penBrain, 2023-04-17:
Could one give a link to a run with such failure?

0penBrain, 2023-04-23:
@berniev any example to give?

chennes, 2023-04-23:
Here is a PR with a failing test: https://github.com/FreeCAD/FreeCAD/pull/9367

berniev, 2023-04-23:
Creating a failing test is really really simple. I don't understand the complexity being expressed.

0penBrain, 2023-04-26:
> Here is a PR with a failing test: #9367

Thanks. I just pushed a PR. When CI ends, I'll merge and ping you if you can rebase the demonstrating PR just to check it's OK now. ;)

0penBrain, 2023-04-26:
@chennes I just merged. Could you please rebase and push #9367 please ?

Sorry BTW, I probably created merge conflicts with your CI PR... :confused:

commit c77b2f33 (2023-04-26): CI: fix failing unit test reported as passed in summary, fixes #9170