FreeCAD/FreeCAD #9177: [Issue] Geometric constraints on Arcs are displayed away of supports

FlachyJoe, 2023-04-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32540 (Git)
Build type: Release
Branch: master
Hash: 07cf291516b8cf52b6c6c8d6dd2ed0ba84b2117c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

Sketcher

### Issue description

When inserting a geometric constraint (perpendicular, tangency or pointOnObject) with an arc as support, the icon is displayed as the supports intersection but it can be away of the arc ends.

### Anything else?

Current view:
[image]

I propose to add an helper below the arc with the constraints color:
[image]

You can also see future issue with arc to something distance: https://github.com/FreeCAD/FreeCAD/pull/9166

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-04-04:
@0penBrain @wwmayer @abdullahtahiriyo Is my idea feasible? If so, can one explain me the constraints display?

luzpaz, 2023-04-10:
bumping to top for feedback

0penBrain, 2023-04-10:
And for doing it, I don't think it's about constraint display but about geometry display. With a mobile now but I can have a look later how spline comb is rendered. I just remember there is a Coin Manager for this.

abdullahtahiriyo, 2023-05-01:
+1 for OpenBrain.

We can look if there is a better heuristic for the position of the constraint.

I we want to provide a visual aid, it should be in the information layer (in one of the coin managers as it is the case of the bspline).

Visual aids should not create new geometry. The limit to create support geometry, today, is when interaction is necessary. Creating new geometry slows the sketches, and it creates lots of other management problems, as the visual aid is tied (or should reasonably be expected to be tied) to the geometry. For example, it needs be deleted together (as internal alignment geometry is). As the circle and the arc will overlap, another concern is that of selecting the right one (or inadvertently selecting the wrong one). These issues do not happen if it is a mere visual aid.

I will nevertheless look into the PR when I get there...

FlachyJoe, 2023-05-01:
@abdullahtahiriyo #9257 implements an aid "in the information layer" i.e. in the same way the bspline comb is.

AjinkyaDahale, 2023-05-06:
I would like it to be only visible for certain cases. It may not be the most relevant here, but B-spline combs tend to clutter the entire scene, so I feel it'd be better to set it/have a setting such that only selected/pre-selected curves have their overlays shown. This could be a separate issue though.

FlachyJoe, 2023-05-06:
@AjinkyaDahale In the arc case the overlay is needed to see the constraints owner. In my opinion it can't be shown only when the arc is selected, in a pinch when the constraint is.

AjinkyaDahale, 2023-05-07:
> In my opinion it can't be shown only when the arc is selected, in a pinch when the constraint is.

When the constraint is selected is a good option.

closed 2023-05-07 by abdullahtahiriyo

abdullahtahiriyo, 2023-05-07:
Actually, I think you both are right. 

The feature helps better identify the constraint owner and the feature clutters the view. What takes priority over what, I think it is a personal opinion.

Therefore, it may be reasonable to provide two modes based on user preferences.

Does that make sense to you?

abdullahtahiriyo, 2023-05-07:
For clarity, such undertaking will not make it into 0.21 release.

PR #9257 (2023-05-07): Sketcher: add a circle overlay for arcs
Fix #9177

The circle overlay is enabled by default and removable by setting Mod/Sketcher/General/ArcCircleHelperVisible preference to False.

[image]
[image]