FreeCAD/FreeCAD #9198: [Problem] Segfault in updateGeometryColor

kenohassler, 2023-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

There is a nullptr dereference in `SketcherGui::EditModeGeometryCoinManager::updateGeometryColor`.
This is reproducible in the Release version (translated with Rosetta), in the weekly arm64 build (31483) and in the weekly x86_64 build (32532).

### How to reproduce
I created a simple sketch (rectangle), closed it and created a new spreadsheet. After entering text in cell A1 and an alias for cell B1, I tried to go back to the sketch. Then FreeCAD crashed.

### Anything else?

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [33054]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        Exited process [33048]
User ID:               501

Date/Time:             2023-04-06 11:13:32.6767 +0200
OS Version:            macOS 13.3 (22E252)
Report Version:        12
Anonymous UUID:        F97A647C-B406-C671-F6F5-8D20A34DBD7E

Sleep/Wake UUID:       6FA00482-1B03-44B6-94D9-5AA9CC7675DD

Time Awake Since Boot: 120000 seconds
Time Since Wake:       2188 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [33054]

VM Region Info: 0 is not in any region.  Bytes before following region: 4339122176
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102a1c000-102a23000    [   28K] r-x/r-x SM=COW  ...s/bin/freecad

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   SketcherGui.so                	       0x182bf3886 SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&, bool) + 774
1   SketcherGui.so                	       0x182befc89 SketcherGui::EditModeCoinManager::updateColor(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&) + 57
2   SketcherGui.so                	       0x182befc06 SketcherGui::EditModeCoinManager::updateColor() + 54
3   SketcherGui.so                	       0x182c1a560 SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) + 2576
4   libFreeCADGui.dylib           	       0x10c5ceefa Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) + 42
5   libFreeCADGui.dylib           	       0x10c359ab0 boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>::dereference() const + 80
6   libFreeCADGui.dylib           	       0x10c3595d1 boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) + 673
7   libFreeCADGui.dylib           	       0x10c5d5b96 Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) + 566
8   libFreeCADGui.dylib           	       0x10c359ab0 boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>::dereference() const + 80
9   libFreeCADGui.dylib           	       0x10c3595d1 boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) + 673
10  libFreeCADGui.dylib           	       0x10c5d438a Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) + 362
11  libFreeCADGui.dylib           	       0x10c5dcaad Gui::SelectionSingleton::addSelections(char const*, char const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&) + 1629
12  SketcherGui.so                	       0x182b7c7e2 CmdSketcherSelectElementsWithDoFs::activated(int) + 1634
13  libFreeCADGui.dylib           	       0x10c193a55 Gui::Command::_invoke(int, bool) + 613
14  libFreeCADGui.dylib           	       0x10c1937b6 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 342
15  SketcherGui.so                	       0x182bce700 SketcherGui::TaskSketcherMessages::on_labelConstrainStatusLink_linkClicked(QString const&) + 496
16  SketcherGui.so                	       0x182bcea48 SketcherGui::TaskSketcherMessages::qt_metacall(QMetaObject::Call, int, void**) + 88
17  libQt5Core.5.15.4.dylib       	       0x1105a8550 0x11045c000 + 1361232

...
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kenohassler, 2023-04-06:
Thanks for looking into this @xtemp09! I investigated a little further and set up two Virtual Machines, one with macOS 12 (Monterey) and one with macOS 13 (Ventura).
On the Ventura machine, the bug is indeed reproducible, so I think it's not my local setup. On Monterey, however, it is *not*.
It came to my mind that Apple changed the semantics of the `free` function in Ventura, such that memory is now zeroed after being free'd. Is this perhaps a use-after-free bug?

luzpaz, 2023-10-11:
@wwmayer it seems @kenohassler has diagnosed this as a use-after-free pointer issue (https://github.com/FreeCAD/FreeCAD/issues/9198#issuecomment-1499213894). Could you weigh-in ?

wwmayer, 2023-10-11:
Not reproducible for me.

luzpaz, 2023-10-12:
wmayer, did you test on macos (about info) ?

@chennes are you able to reproduce?

wwmayer, 2023-10-12:
I don't have macOS.

luzpaz, 2023-10-12:
@JohnOCFII do you mind testing on stable and development ?

chennes, 2023-10-12:
I cannot reproduce on:
```
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34650 (Git)
Build type: Release
Branch: main
Hash: 4f1fcbd03b703fb76886c1bbfec172167ea48aef
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.5.3
```

JohnOCFII, 2023-10-12:
> @JohnOCFII do you mind testing on stable and development ?

Can not duplicate issue on my M1/arm MacBook Pro on either:

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
``` 

nor

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Release
Branch: master
Hash: e9187412d9b3b372ee4c1d7e2fc7a0bf08a2e647
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

kenohassler, 2023-10-12:
Hi y'all,
thanks for looking into this! Very much appreciated. I just tried it again with the current release on my machine, and it still crashes in the same function.

```
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

I'll try to make this more reproducible! Take the attached file. It was produced as described above. Open it, and open the spreadsheet. Then, open the sketch. For me, this reproducibly crashes (tried 3 times in a row). Please, let me know if this helps, and if there is something else I can provide.
[cube2.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12886275/cube2.FCStd.zip)

JohnOCFII, 2023-10-14:
> It was produced as described above. Open it, and open the spreadsheet. Then, open the sketch. For me, this reproducibly crashes (tried 3 times in a row).

Thanks for replying and offering the test file.  My system doesn't crash with this either.  There are a few Mac-specific issues out there, and not sure what the differences are.  What machine type do you have? It looks like your initial report was with macOS 10.16, and your most current test was with macOS 14.0.  How much memory do you have in the machine?  Do you use external monitors or mice/pointing devices?  All my tests have been with a 2021 MacBook Pro with 32GB RAM and just using the built-in TrackPad.  My machine is pretty clean.

kenohassler, 2023-10-14:
I actually have the same machine, also 32 gigs of RAM. Both reports come from this MBP, I just upgraded macOS recently. The recent test was without any devices attached, the first one was with an external monitor, I believe. My machine isn't exactly clean, I have a ton of brew packages installed - but at least in April, I could reproduce in a fresh VM, which should exclude this as an error source?

wwmayer, 2023-10-14:
When looking at function `EditModeGeometryCoinManager::updateGeometryColor` then from line 171 on it is not guaranteed that the access to several C++ vectors and C arrays is safe. It's probably best to refactor it completely to fix all this mess.

kenohassler, 2023-10-15:
Small update: I tested again with the most recent weekly build for arm64 (0.22.0devR34651). Here, I get a different SegFault (trace attached). Is it feasible to compile FreeCAD with ASAN on Linux to find these kinds of bugs without a Mac?

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [45966]
Path:                  /Applications/FreeCAD weekly.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [45954]
User ID:               501

Date/Time:             2023-10-15 18:48:18.4106 +0200
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Anonymous UUID:        F97A647C-B406-C671-F6F5-8D20A34DBD7E

Sleep/Wake UUID:       3380F90A-8A7D-429D-A950-9CA18DAFA12C

Time Awake Since Boot: 270000 seconds
Time Since Wake:       6775 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000001bd
Exception Codes:       0x0000000000000001, 0x00000000000001bd

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [45966]

External Modification Warnings:
Debugger attached to process.

VM Region Info: 0x1bd is not in any region.  Bytes before following region: 4366761539
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104478000-104498000    [  128K] r-x/r-x SM=COW  ...s/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libCoin.4.0.0.dylib           	       0x108fd0c68 SoRenderManager::getCamera() const + 4
1   SketcherGui.so                	       0x2a929e3d8 SketcherGui::ViewProviderSketch::camSensCB(void*, SoSensor*) + 24
2   libCoin.4.0.0.dylib           	       0x109002664 SoDataSensor::trigger() + 20
3   libCoin.4.0.0.dylib           	       0x109004704 SoSensorManager::processDelayQueue(int) + 404
4   libFreeCADGui.dylib           	       0x106265504 SIM::Coin3D::Quarter::SensorManager::idleTimeout() + 36
5   libQt5Core.5.15.8.dylib       	       0x1081e9424 0x1080a4000 + 1332260
6   libQt5Core.5.15.8.dylib       	       0x1081cc71c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
7   CoreFoundation                	       0x189e0fcbc __CFRunLoopDoSource0 + 176
8   CoreFoundation                	       0x189e0fa8c __CFRunLoopDoSources0 + 340
9   CoreFoundation                	       0x189e0e61c __CFRunLoopRun + 828
10  CoreFoundation                	       0x189e0dc2c CFRunLoopRunSpecific + 608
11  HIToolbox                     	       0x194366448 RunCurrentEventLoopInMode + 292
12  HIToolbox                     	       0x1943660d8 ReceiveNextEventCommon + 220
13  HIToolbox                     	       0x194365fdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
14  AppKit                        	       0x18d5e8f90 _DPSNextEvent + 660
15  AppKit                        	       0x18ddbcb94 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
16  AppKit                        	       0x18d5dc43c -[NSApplication run] + 476
17  libqcocoa.dylib               	       0x109d18a64 0x109cec000 + 182884
18  freecad                       	       0x10447c568 main + 5064
19  dyld                          	       0x1899b9058 start + 2224
```

wwmayer, 2023-10-16:
>  Is it feasible to compile FreeCAD with ASAN on Linux to find these kinds of bugs without a Mac?

I have an ASAN build on Linux but for me it doesn't crash.

kenohassler, 2023-10-22:
Ok, damn. Perhaps some sublte differences in behaviour of macOS vs Linux... I'll try to compile a (ASAN?) debug build when I find time and let you know where exactly things happen 😄

xtemp09, 2024-01-13:
@kenohassler, if you look at the backtrace you provided, it similar to the one I encountered in #11954.

SIM::Coin3D::Quarter::SensorManager::idleTimeout → SoSensorManager::processDelayQueue(int) → SoDataSensor::trigger() → SketcherGui::ViewProviderSketch::camSensCB

chennes, 2024-01-13:
For reference (and as a warning to others who think this might be a quick fix...) `EditModeGeometryCoinManager::updateGeometryColor` is 350 lines long, and Clang-tidy computes its "cognitive complexity" as 181 (25 is widely considered bad enough to warrant refactoring). So it is definitely due, and will be a bit of a project in its own right.

maxwxyz, 2024-10-07:
Is this still an issue in the latest RC / dev version?

closed 2024-10-18 by kenohassler