FreeCAD/FreeCAD #9203: NOT RESPONDING error when working in sketch and trying to sketch my item

Jasonv310, 2023-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * ArchTextures
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Design456 0.0.1
  * fasteners 0.4.56
  * MeshRemodel 1.8919.0
  * sheetmetal 0.2.63
  * slic3r-tools
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

I am working on a sketch of my item and I have spent days on it and I am now constantly getting a "not responding" error and it has slowed the software down alot, in which makes for a very painful experience with the program. Could you please help ASAP?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-04-06:
Have you tried the [latest weekly build](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)?

howie-j, 2023-04-06:
Can you share your file?

Jasonv310, 2023-04-06:
I am not sure how to do that. I am very new to this program
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

xtemp09, 2023-04-06:
Are not sure about downloading the latest weekly build, using the backup or sharing your file?

Also, you can try a virtual machine. Perhaps, on Ubuntu everything is working fine.

Jasonv310, 2023-04-06:
I am not sure how to download the weekly builds. I am not super good at the computer stuff yet especially with FreeCAD and Python, etc...
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

xtemp09, 2023-04-06:
Ok. Are sure how to rename *.FCStd1, *.FCStd2 into *.FCStd?

Jasonv310, 2023-04-06:
unfortunately I am not sure where to code that into or what I am renaming. Sorry. 
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

xtemp09, 2023-04-06:
The directory where you keep your file contains files ***.FCStd1**, ***.FCStd2** and other ones. They are _backups_ for the project which has the name ***.FCStd**.

Take a look at the directory and rename ***.FCStd1** or ***.FCStd2** into ***.FCStd**. You may consider the date and the time when they were created and choose the most recent or the oldest.

Jasonv310, 2023-04-06:
How and where do i find the "directory"
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

Jasonv310, 2023-04-06:
this program is awful, I have put days into this sketch and everything is just "not responding" It makes me want to throw my computer outside and give up. 
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

Jasonv310, 2023-04-06:
It seems to just be getting worse now

howie-j, 2023-04-06:
As said, please share your file. This is probably a user error, not a freecad fault.

_Days_ into a single sketch? Sketches should be as small and lightweight as possible, and contain only what is necessary to get the job done.

Jasonv310, 2023-04-06:
would you mind helping me figure out how to share my file and I would be happy to
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

xtemp09, 2023-04-06:
Archive it with zip or any other program and drag-and-drop it into the text field here.

Jasonv310, 2023-04-06:
I have been trying to sketch my image that i want, so I am able to extrude it onto another object and so on and even have a cutout of it on an object. and it was my understanding that i had to trace my picture first?
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

Jasonv310, 2023-04-06:
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

xtemp09, 2023-04-06:
I don't think that sending file over e-mail works here. I meant attaching them on the text field on _github.com_.

xtemp09, 2023-04-06:
If you are using e-mail, try sending it to xtemp09@gmail.com

Jasonv310, 2023-04-06:
I just sent it to you. 
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

Jasonv310, 2023-04-06:
I just wanted to make sure that you got what I sent? I greatly appreciate your help with fixing my issues if possible. 
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.***

luzpaz, 2023-04-07:
1. Take the FreeCAD file and zip it using a 3rd party zip program  
2. Then open your browser to open https://github.com/FreeCAD/FreeCAD/issues/9203 
3. Drag and drop the zipped file in to the text field and press 'Comment'

xtemp09, 2023-04-07:
Jasonv310's file: [Guardian Sketch (3).zip](https://github.com/FreeCAD/FreeCAD/files/11175560/Guardian.Sketch.3.zip)

howie-j, 2023-04-07:
That is unfortunately a user error, or at least trying to do something the sketcher is not designed to do.

One option is to split the logo into multiple sketches, for example one for the outline, one for the middle part, one for the inner outline etc. That will make each sketch much simpler and faster to work with. If you do this, please scale your image to the final size before starting, and only use the "polyline" tool to add geometry. Your sketch has multiple open wires. Also, do not add dimensional constraints, that will slow the sketch down, keep the sketch unconstrained (but with lines connected!)

Another option is to import the svg directly to draft, and automatically convert the wires to sketch. I had trouble importing your svg file however.

Third option is to convert your logo to dxf, and use draft dxf to sketch.

I would also recommend asking on the forum, you will probably get faster replies there: [forum.freecad.org](forum.freecad.org)

luzpaz, 2023-04-07:
Closing ticket

closed 2023-04-07 by luzpaz

Jasonv310, 2023-04-07:
thank you all for the help

Jasonv310, 2023-04-07:
ok a couple more questions:
1. What is the easiest way to scale my object that I want to sketch2. how do I get rid of all the points that show up on my sketch3. What do I need to do to prepare my made sketch, so I can put it on top of a 6mmX303mm cylinder and then do an imprint of it on the object so it is attached to that object. 
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.*** 

 
 

Closed #9203 as not planned.

xtemp09, 2023-04-07:
If I were you, I would have imported the logo into _Inkscape_, inverted it and instead of extruding I would have projected it and cut out the projection. From my point of view, this approach is simpler.

[This video](https://www.youtube.com/watch?v=bMhMiPaqH3U) (and, actually, the channel) might help you out.