FreeCAD/FreeCAD #9204: Not Resonding error when using sketcher and sketching my picture

Jasonv310, 2023-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * ArchTextures
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Design456 0.0.1
  * fasteners 0.4.56
  * MeshRemodel 1.8919.0
  * sheetmetal 0.2.63
  * slic3r-tools
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

After spending days on sketching my picture, I am now getting the "not responding" error and extremely slow responses in which makes for a miserable experience with this program. Can you please help ASAP? 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-04-06:
Can you share your file?

howie-j, 2023-04-06:
Please don't spam/double post: https://github.com/FreeCAD/FreeCAD/issues/9203

closed 2023-04-07 by luzpaz