FreeCAD/FreeCAD #9205: Problem:  Default settings make it hard to select objects in the 3D window

sliptonic, 2023-04-06:


howie-j, 2023-04-06:
Also related to this issue; increasing the pick radius makes it even more difficult to select internal/concave edges (inside edge of a box for example).

PaddleStroke, 2023-04-12:
Also when you increase the pick radius, then a line that is on front will have priority from another line even if the cursor is closer to this other line. On picture the highlighted line is hard to select.
[image]

PaddleStroke, 2023-04-12:
Regarding improving the selection of things I had a look. Here are my notes for reference. Please note that I'm not 100% sure about those.
 
- It seems that the selection rely on `View3DInventorViewer::getPointOnRay`.
This function seems to be getting a single `SoPickedPoint `with `SoRayPickAction.getPickedPoint()`.
- This SoPickedPoint seems to be processed in the different workbenches. For example in sketcher it is processed in : `EditModeCoinManager::detectPreselection`.

So the selection seems to rely on SoRayPickAction. From [its doc](https://www.openinventor.com/reference-manuals/NewRefMan1030/RefManJava/com/openinventor/inventor/actions/SoRayPickAction.html) we can read that it is also possible to get a list of points hit by the ray and not just one : 

> Callers can request the action to compute all intersections along the ray (sorted closest to farthest) by setting the pickAll flag to true. By default, the action computes only the closest intersection.

So this makes me think that by using `getPickedPoint` we only get the item that is the closest to the camera.

So I think we should instead get the list of points, then process this list in each workbenches to determine which feature to select. Then we can set rules like 'Points are prioritary on edges' and 'edges are prioritary on faces'.

If someone can confirm or deny or comment my assumptions ?

sliptonic, 2023-04-12:
I haven't had the problem with internal edges.  Maybe I'm looking at it wrong.  
But pick radius, has a huge affect on the ability to select sketcher geometries and constraint icons. With a large pick radius I almost cannot pick the tangency icon but can easily select the vertex.  

With a small radius I can easily get the tangency or the line but not the vertex.  
[image]

MadMark13, 2023-04-13:
From my experience much of the pick behavior on icons and text seem to be on the painted pixels of the particular icon or text you are selecting, where I'd prefer a bounding box around said text or icon or even the centroid of it plus a select radius.  
This results in frustration trying to double-click on a dimension, but if my pick radius is small, I have to make sure my pointer is on the letter strokes to do that.  More than once I've double-clicked in the middle of a zero to, well, zero effect.

Like PaddleStroke said, but include the constraint icons to the class containing points for the selection priority rules.
Could do it with a icon/text centroid plus pickable radius and pick radius overlap, once they overlap it's a potential pick, then the rules would come into play on the click event.

The hand/eye coordination is 2D, so pick behavior, ideally, should follow that.  It would solve the foreground line picked over the slightly background line that is closer to pick center.  If far background geometry is desired to be pickable, in the case of wireframe, or transparent geometry, those could still be picked, but maybe using an ALT to a stack selection dialog.

Hunting to click this much makes me pine for Catia, and that's just sad.

dgrabulosa, 2023-04-13:
[Related post](https://forum.freecad.org/viewtopic.php?t=72047) 
Some pics for reference difficult selectable edges.
Diferent behavior depending from what edge side is made the selection.
[image]
[image]

danielpeterson, 2023-08-29:
I have spent the morning investigating this. I am unfamiliar with the Freecad codebase but here is what I have found.

For the 3d view this behavior is handled by the following function:
```cpp
SoFCUnifiedSelection::getPickedList(SoHandleEventAction* action, bool singlePick) const
```

The "bug" with erratic highlights seems to stem from the following logic:

```cpp
    for(auto it=ret.begin()+1;it!=ret.end();++it) {

        if ((cur_prio > picked_prio) && picked_pt.equals(cur_pt, 0.01f)) {
            itPicked = it;
            picked_prio = cur_prio;
        }
    }
```

Specifically it is this logic that is causing the problem:
```cpp
if ((cur_prio > picked_prio) && picked_pt.equals(cur_pt, 0.01f))
```

It selects the highest priority item within a small region around the cursor. Otherwise it selects the item closest to the camera. Priority is set in the adjacent getPriority function:

```cpp
int SoFCUnifiedSelection::getPriority(const SoPickedPoint* p)
{
    const SoDetail* detail = p->getDetail();
    if (!detail)
        return 0;
    if (detail->isOfType(SoFaceDetail::getClassTypeId()))
        return 1;
    if (detail->isOfType(SoLineDetail::getClassTypeId()))
        return 2;
    if (detail->isOfType(SoPointDetail::getClassTypeId()))
        return 3;
    return 0;
}
```

I think the behavior that the user expects is the following:

* Only select from visible geometry
* Prioritize points over lines over faces
* If two features of the same priority (two lines for example) are both within the pick radius, select the one that is the closest to the tip of the cursor as measured in the view plane.

I will se if I can figure out the following:
* How does one find the view-plane projection of the cursor and picked points from within the getPickedList function.
* How to determine the visibility of the picked features from within the getPickedList function.

If anyone has suggestions, please let me know.

obelisk79, 2023-08-29:
Daniel, @wwmayer might be the most knowledgeable one to answer your questions.

wwmayer, 2023-09-18:
Coin3D is able to find also edges of concave areas but the problems is caused by the checks afterwards that ignores a correctly detected edge.

After doing some tests it appeared it's just the second part of the if-clause that causes problems:
`picked_pt.equals(cur_pt, 0.01f)`

This tolerance of 0.01 works OK for convex areas but is by far too strict for concave areas. You can imagine that for a small angle between the adjacent faces of a concave edge there is a relative high distance between the found pick points. The smaller this angle is the higher are the distances between these points.

After doing some more tests we should work with a tolerance of 0.2.

Tofandel, 2023-12-09:
Is the issue fixed in a nightly? I'm also having a super hard time selecting lines on a 4K screen and changing the pick radius doesn't help

dzid26, 2024-04-07:
I don't have a problem selecting anything (vertices, lines on concave shapes) so far, but I discovered there is Pick Object list available in Selection List panel to help with the selecting anything within the picker radius:
[image]
(Requires double clicking to add to current selection)

The issue is this panel is doesn't work during an editing tasks.

closed 2025-01-19 by github-actions[bot]