FreeCAD/FreeCAD #9206: [Problem] Sketcher Marker Size setting is misplaced

sliptonic, 2023-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Problem description

The General / 3D View / Rendering section has a setting for 'Marker size'.  This setting controls the size of vertices in the Sketcher workbench.  

[image]

This setting should be moved to the Sketcher Group in the Display Tab
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-04-06:
I believe the reason that _Marker Size_ is in Display>3D View is due to the number of workbenches that specific _View_ parameter is used:
Fem, Mesh, PartDesign, Path, Robot and Sketcher so the tooltip is misleading to say the least so change the tooltip but it shouldn't be moved to a specific Wb when it's used elsewhere.

benj5378, 2023-07-19:
> Fem, Mesh, PartDesign, Path, Robot and Sketcher so the tooltip is misleading to say the least so change the tooltip but it shouldn't be moved to a specific Wb when it's used elsewhere.

@sliptonic any comments on this? Should we just change the tooltip?

howie-j, 2023-08-13:
I can at least verify that MarkerSize changes TechDraw vertices, so perhaps this setting is correctly placed after all.
Updated tooltip: https://github.com/FreeCAD/FreeCAD/pull/10215

closed 2023-08-16 by luzpaz