FreeCAD/FreeCAD #9208: [Issue] Sketcher Toolbar Configuration not Saved

howie-j, 2023-04-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Fedora Linux 37 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32764 (Git)
Build type: Release
Branch: master
Hash: 8ea5d9bac10698a9ae31735ad45a63688a6be58b
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

_Sketcher edit mode toolbars_ position and visibility is not saved to user.cfg after modification.

Steps to reproduce:

- Switch to sketcher
- New sketch, open edit mode
- Rearrange toolbars
- Hide toolbars
- Close FreeCAD

When FreeCAD is restarted all sketcher toolbars are restored to default visibility and position.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-04-06:
Reading from a previously configured user.cfg works, but writing the current setup does not.

Added a user.cfg file for testing:
[user_cfg.zip](https://github.com/FreeCAD/FreeCAD/files/11174090/user_cfg.zip)

Reading toolbar positions works, but changes are not saved.

Syres916, 2023-04-09:
From some of my testing it appears to be https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/ToolBarManager.cpp#L347 section that is causing the regression you are seeing and it's the Sketcher Edit Mode that possibly appears to require an additional condition to not be subject to this Ignore function. @wwmayer am I way off the mark here?

adrianinsaval, 2023-04-14:
@PaddleStroke may be related to the auto hiding?

abdullahtahiriyo, 2023-05-28:
@Syres916 

The position of the toolbars does get saved on my side. The visibility of the toolbars does not (if I hide one toolbar, it will appear again if I close FC and reopen it) . Do you have the same behaviour?

abdullahtahiriyo, 2023-05-28:
@wwmayer 

I am looking at this one.

If I hide the constraints toolbar in sketcher edit mode, when I leave edit mode, Sketcher/Gui/Workbench.cpp, Workbench::leaveEditMode() gets called. There is a comment that the sate must be saved first. There is a call to ```Gui::ToolBarManager::getInstance()->saveState();```.  Eventually the constraint bar gets iterated and the ignoreSave lambda gets executed:
https://github.com/FreeCAD/FreeCAD/blob/21223c82270cc8440fe9d873a774e7186a206324/src/Gui/ToolBarManager.cpp#L347

While the toolbar is not visible, the toggleaction isVisible, the comment indicates this is not being controlled by the application, so "do not ignore".

Because for the constraint bar the style is ToolBarItem::HideStyle::FORCE_HIDE, saving gets ignored.

I do not fully grasp how the mechanism is intended to work.

Is sketcher supposed to handle the saving of these toolbars manually? How?

wwmayer, 2023-05-28:
> The position of the toolbars does get saved on my side. The visibility of the toolbars does not (if I hide one toolbar, it will appear again if I close FC and reopen it) .

Same behaviour here. You don't have to restart FC but switching to another wb and back or leaving edit mode and re-entering shows the same issue.

> Because for the constraint bar the style is ToolBarItem::HideStyle::FORCE_HIDE, saving gets ignored.

It's on purpose not to save visibility of toolbars that are managed by the application because otherwise you will end up in a situation where all relevant toolbars are missing. The changes were made with PR #9150 that fixed issue #9135.

But all these kind of problems started with c9aadeec4ce7b18ab

abdullahtahiriyo, 2023-05-28:
Thanks Werner. 

I was missing the last commit where all originated.

I will then look at it from a different perspective.

abdullahtahiriyo, 2023-05-29:
@Syres916 
@howie-j 

Feel free to give it a run to the PR above and let me know if it does not meet expectations.

howie-j, 2023-05-29:
Thanks for your work @abdullahtahiriyo, just tested pr #9694. It fixes the problem as described, and it seems to save toolbar states even if you close freecad in sketch edit mode (sketcher leaves edit mode when you get the promt to save file).

However i noticed one issue: on a clean install (no user.cfg exists), the default toolbar visibility is not written to user.cfg, meaning you have no toolbars as default:
[image]

abdullahtahiriyo, 2023-05-30:
@howie-j 
@Syres916 

Thank you. Yes, indeed the edit-mode toolbars defaulted to not visible when entering sketch edit mode.

I have pushed a fix.

Please give it a run and let me know if there are other outstanding issues.

abdullahtahiriyo, 2023-05-30:
@adrianinsaval 

I saw your comment in the PR.

It will save state every time edit mode is left with the "Sketcher WB" workbench being the active WB. No matter how. Closing FC included. Of course, if you kill the process (or FreeCAD segfaults), it does not quite qualify as "leaving edit mode" and then it will not be saved. But that is probably beyond user expectations.

Give it a run if you have time and let me know if it misbehaves.

Syres916, 2023-05-30:
Thanks for the efforts @abdullahtahiriyo , I've tried the PR using my normal workflow of _Sketcher_, _Part_, _Draft_ along with some macros and it appears to be fixed for me. I'll keep working with it for the next few days but I for one am happy for the PR to be merged.

closed 2023-05-30 by abdullahtahiriyo